smlinux/rtl8723de #1: How to build it on kernel 4.13 ubuntu?

anshulgoyal15, 2017-12-21:
No info on building provided

anshulgoyal15, 2017-12-21:
can you tell me how to compile it on ubuntu

anshulgoyal15, 2017-12-21:
problem in building
`make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/4.13.0-19-generic/build M=/home/anshul/rtl8723de  modules
make[1]: Entering directory '/usr/src/linux-headers-4.13.0-19-generic'
  CC [M]  /home/anshul/rtl8723de/core/rtw_cmd.o
In file included from /home/anshul/rtl8723de/include/drv_types.h:32:0,
                 from /home/anshul/rtl8723de/core/rtw_cmd.c:22:
/home/anshul/rtl8723de/include/osdep_service.h: In function ‘thread_enter’:
/home/anshul/rtl8723de/include/osdep_service.h:369:2: error: implicit declaration of function ‘allow_signal’; did you mean ‘do_signal’? [-Werror=implicit-function-declaration]
  allow_signal(SIGTERM);
  ^~~~~~~~~~~~
  do_signal
/home/anshul/rtl8723de/include/osdep_service.h: In function ‘flush_signals_thread’:
/home/anshul/rtl8723de/include/osdep_service.h:391:6: error: implicit declaration of function ‘signal_pending’; did you mean ‘timer_pending’? [-Werror=implicit-function-declaration]
  if (signal_pending(current))
      ^~~~~~~~~~~~~~
      timer_pending
/home/anshul/rtl8723de/include/osdep_service.h:392:3: error: implicit declaration of function ‘flush_signals’; did you mean ‘do_signal’? [-Werror=implicit-function-declaration]
   flush_signals(current);
   ^~~~~~~~~~~~~
   do_signal
cc1: some warnings being treated as errors
scripts/Makefile.build:302: recipe for target '/home/anshul/rtl8723de/core/rtw_cmd.o' failed
make[2]: *** [/home/anshul/rtl8723de/core/rtw_cmd.o] Error 1
Makefile:1546: recipe for target '_module_/home/anshul/rtl8723de' failed
make[1]: *** [_module_/home/anshul/rtl8723de] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-4.13.0-19-generic'
Makefile:1884: recipe for target 'modules' failed
make: *** [modules] Error 2`

anshulgoyal15, 2017-12-21:
build problem on linux kernel 4.13.09
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/4.13.0-19-generic/build M=/home/anshul/rtl8723de  modules
make[1]: Entering directory '/usr/src/linux-headers-4.13.0-19-generic'
  CC [M]  /home/anshul/rtl8723de/core/rtw_cmd.o
In file included from /home/anshul/rtl8723de/include/drv_types.h:32:0,
                 from /home/anshul/rtl8723de/core/rtw_cmd.c:22:
/home/anshul/rtl8723de/include/osdep_service.h: In function ‘thread_enter’:
/home/anshul/rtl8723de/include/osdep_service.h:369:2: error: implicit declaration of function ‘allow_signal’; did you mean ‘do_signal’? [-Werror=implicit-function-declaration]
  allow_signal(SIGTERM);
  ^~~~~~~~~~~~
  do_signal
/home/anshul/rtl8723de/include/osdep_service.h: In function ‘flush_signals_thread’:
/home/anshul/rtl8723de/include/osdep_service.h:391:6: error: implicit declaration of function ‘signal_pending’; did you mean ‘timer_pending’? [-Werror=implicit-function-declaration]
  if (signal_pending(current))
      ^~~~~~~~~~~~~~
      timer_pending
/home/anshul/rtl8723de/include/osdep_service.h:392:3: error: implicit declaration of function ‘flush_signals’; did you mean ‘do_signal’? [-Werror=implicit-function-declaration]
   flush_signals(current);
   ^~~~~~~~~~~~~
   do_signal
cc1: some warnings being treated as errors
scripts/Makefile.build:302: recipe for target '/home/anshul/rtl8723de/core/rtw_cmd.o' failed
make[2]: *** [/home/anshul/rtl8723de/core/rtw_cmd.o] Error 1
Makefile:1546: recipe for target '_module_/home/anshul/rtl8723de' failed
make[1]: *** [_module_/home/anshul/rtl8723de] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-4.13.0-19-generic'
Makefile:1884: recipe for target 'modules' failed
make: *** [modules] Error 2
anshul@coding-geek-HP-Laptop-15-bs0xx:~/rtl8723de$ make -Werror 
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/4.13.0-19-generic/build M=/home/anshul/rtl8723de  modules
make[1]: Entering directory '/usr/src/linux-headers-4.13.0-19-generic'
  CC [M]  /home/anshul/rtl8723de/core/rtw_cmd.o
In file included from /home/anshul/rtl8723de/include/drv_types.h:32:0,
                 from /home/anshul/rtl8723de/core/rtw_cmd.c:22:
/home/anshul/rtl8723de/include/osdep_service.h: In function ‘thread_enter’:
/home/anshul/rtl8723de/include/osdep_service.h:369:2: error: implicit declaration of function ‘allow_signal’; did you mean ‘do_signal’? [-Werror=implicit-function-declaration]
  allow_signal(SIGTERM);
  ^~~~~~~~~~~~
  do_signal
/home/anshul/rtl8723de/include/osdep_service.h: In function ‘flush_signals_thread’:
/home/anshul/rtl8723de/include/osdep_service.h:391:6: error: implicit declaration of function ‘signal_pending’; did you mean ‘timer_pending’? [-Werror=implicit-function-declaration]
  if (signal_pending(current))
      ^~~~~~~~~~~~~~
      timer_pending
/home/anshul/rtl8723de/include/osdep_service.h:392:3: error: implicit declaration of function ‘flush_signals’; did you mean ‘do_signal’? [-Werror=implicit-function-declaration]
   flush_signals(current);
   ^~~~~~~~~~~~~
   do_signal
cc1: some warnings being treated as errors
scripts/Makefile.build:302: recipe for target '/home/anshul/rtl8723de/core/rtw_cmd.o' failed
make[2]: *** [/home/anshul/rtl8723de/core/rtw_cmd.o] Error 1
Makefile:1546: recipe for target '_module_/home/anshul/rtl8723de' failed
make[1]: *** [_module_/home/anshul/rtl8723de] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-4.13.0-19-generic'
Makefile:1884: recipe for target 'modules' failed
make: *** [modules] Error 2

smlinux, 2017-12-21:
Fixed. Commit d7c05e91eb6039263ffe34349cfc2c8dc2879f54

closed 2017-12-21 by smlinux