smlinux/rtl8723de #2: fatal error: linux/sched/signal.h: No such file or directory

michaelkrukov, 2017-12-21:
Lubuntu 17.04, Kernel 4.10.0-42

Result of `make` command: 

```
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/4.10.0-42-generic/build M=/home/michael/Downloads/rtl8723de  modules
make[1]: Entering directory '/usr/src/linux-headers-4.10.0-42-generic'
  CC [M]  /home/michael/Downloads/rtl8723de/core/rtw_cmd.o
In file included from /home/michael/Downloads/rtl8723de/include/drv_types.h:32:0,
                 from /home/michael/Downloads/rtl8723de/core/rtw_cmd.c:22:
/home/michael/Downloads/rtl8723de/include/osdep_service.h:41:33: fatal error: linux/sched/signal.h: No such file or directory
  #include <linux/sched/signal.h>
                                 ^
compilation terminated.
scripts/Makefile.build:294: recipe for target '/home/michael/Downloads/rtl8723de/core/rtw_cmd.o' failed
make[2]: *** [/home/michael/Downloads/rtl8723de/core/rtw_cmd.o] Error 1
Makefile:1524: recipe for target '_module_/home/michael/Downloads/rtl8723de' failed
make[1]: *** [_module_/home/michael/Downloads/rtl8723de] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-4.10.0-42-generic'
Makefile:1884: recipe for target 'modules' failed
make: *** [modules] Error 2
```

michaelkrukov, 2017-12-21:
Updated to Linux 4.14.0-041400-generic and error went away.
Guide i used: https://www.tecmint.com/upgrade-kernel-in-ubuntu/

smlinux, 2017-12-21:
Now for the kernel version <= 4.10 use the branch 4.10-down, and >=4.11 - branch 4.11-up

closed 2017-12-21 by smlinux