smlinux/rtl8723de #4: Is the Bluetooth for RTL8723DE made to work with related to this driver?

AvirahJ, 2017-12-22:
Hi, 
Thank you for the dirver provided for RTL8723DE wifi module. It works great so far. But, another issue with the new laptops like HP 15-BS576TX/BS000 range and others is, bluetooth driver missing.  My assumption is, WiFi and bluetooth share the same hardware.

jeremyb31, 2017-12-23:
It is the same hardware but the bluetooth requires firmware and it might require changes to the btusb and btrtl modules

AvirahJ, 2017-12-24:
@jeremyb31 So, Will there any solution to get the bluetooth hardware to get to work? Any idea if, the bt firmware will be released by the manufacturer (Realtek).

jeremyb31, 2017-12-24:
If and when it is released by Realtek it should be logged at https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/log/rtl_bt
The strange thing is, is that I remember 1 post where the user said the bluetooth did work on the 8723DE chipset, https://forums.linuxmint.com/viewtopic.php?f=53&t=257254
What does `lsusb` reveal?

AvirahJ, 2017-12-24:
Well, lsusb list this:
`Bus 001 Device 002: ID 0bda:b009 Realtek Semiconductor Corp. `

A verbose lsusb output is here: https://pastebin.com/u48H4X4v 

dmesg complains of lacking firmware blob: 
` kernel: Bluetooth: Core ver 2.22
 kernel: Bluetooth: HCI device and connection manager initialized
 kernel: Bluetooth: HCI socket layer initialized
 kernel: Bluetooth: L2CAP socket layer initialized
 kernel: Bluetooth: SCO socket layer initialized

 kernel: usbcore: registered new interface driver btusb
 kernel: Bluetooth: hci0: rtl: examining hci_ver=08 hci_rev=000d lmp_ver=08 lmp_subver=8723
 kernel: Bluetooth: hci0: rtl: loading rtl_bt/rtl8723b_config.bin
 kernel: AVX2 version of gcm_enc/dec engaged.
 kernel: AES CTR mode by8 optimization enabled
 kernel: bluetooth hci0: Direct firmware load for rtl_bt/rtl8723b_config.bin failed with error -2
 kernel: bluetooth hci0: Falling back to user helper
 kernel: Bluetooth: hci0: rtl: loading rtl_bt/rtl8723b_fw.bin
 kernel: Bluetooth: hci0: rom_version status=0 version=2
 kernel: Bluetooth: hci0: didn't find patch for chip id 2
 kernel: Bluetooth: HCI UART driver ver 2.3
 kernel: Bluetooth: HCI UART protocol H4 registered
 kernel: Bluetooth: HCI UART protocol BCSP registered
 kernel: Bluetooth: HCI UART protocol LL registered
 kernel: Bluetooth: HCI UART protocol ATH3K registered
 kernel: Bluetooth: HCI UART protocol Three-wire (H5) registered
 kernel: Bluetooth: HCI UART protocol Intel registered
 kernel: Bluetooth: HCI UART protocol Broadcom registered
 kernel: Bluetooth: HCI UART protocol QCA registered
 kernel: Bluetooth: HCI UART protocol AG6XX registered
 kernel: Bluetooth: HCI UART protocol Marvell registered
 kernel: 8723de: loading out-of-tree module taints kernel.

 kernel: Bluetooth: BNEP (Ethernet Emulation) ver 1.3
 kernel: Bluetooth: BNEP filters: protocol multicast
 kernel: Bluetooth: BNEP socket layer initialized`

``:~$ hciconfig 
hci0:	Type: Primary  Bus: USB
	BD Address: 00:00:00:00:00:00  ACL MTU: 0:0  SCO MTU: 0:0
	DOWN 
	RX bytes:21 acl:0 sco:0 events:2 errors:0
	TX bytes:6 acl:0 sco:0 commands:2 errors:0

Offtopic: The amount of messages rtl8723de driver module  throws into the kern.log is massive. I wonder, is this normal behaviour?

AvirahJ, 2017-12-24:
I cannot up the device - ` :~$ sudo hciconfig hci0 up`
`Can't init device hci0: Invalid argument (22)`

jeremyb31, 2017-12-24:
You will have to wait for the firmware to get updated for bluetooth to function.  The module might have some debugging level enabled since it is new

AvirahJ, 2017-12-24:
Here is a snapshot of kern.log full of talkative module : https://paste.ubuntu.com/26247041/ Full of messages like: `Dec 24 22:31:13  kernel: RTW: traffic_status_watchdog(wlo1) acqiure wake_lock for 4500 ms(tx:99,rx_unicast:132)`

jeremyb31, 2017-12-24:
`echo "options 8723de rtw_power_mgnt=0 rtw_drv_log_level=0" | sudo tee /etc/modprobe.d/8723de.conf`
That should change some things after a reboot or you could
`sudo modprobe -r rtl8723de && sleep 20 && sudo modprobe rtl8723de`

AvirahJ, 2017-12-25:
@jeremyb31 Well, that module options seems to have little effect.
Messages like below is constantly logged on every session:

:~$ tail  -f /var/log/kern.log
> Dec 25 22:56:13  kernel: RTW: ===>phy_SwChnl8723D: Channel = 12
> Dec 25 22:56:13  kernel: RTW: ===>phy_SpurCalibration_8723D: Channel = 13
> Dec 25 22:56:13  kernel: RTW: ===>phy_SwChnl8723D: Channel = 13
> Dec 25 22:56:13  kernel: RTW: ===>phy_SpurCalibration_8723D: Channel = 4
> Dec 25 22:56:13  kernel: RTW: ===>phy_SwChnl8723D: Channel = 4
> Dec 25 22:56:13  kernel: RTW: SetHwReg8723D(wlo1): [HW_VAR_MACID_WAKEUP] macid=0, org reg_0x4d4=0x00000001
> Dec 25 22:56:13  kernel: RTW: survey done event(7) band:0 for wlo1
> Dec 25 22:56:13  kernel: RTW: rtw_indicate_scan_done(wlo1)
> Dec 25 22:56:13  kernel: RTW: rtw_set_ps_mode(wlo1) Enter 802.11 power save - WIFI-TRAFFIC_IDLE
> Dec 25 22:56:13  kernel: RTW: rtl8723d_set_FwPwrMode_cmd(): FW LPS mode = 2, SmartPS=2
> 
> Dec 25 22:56:36  kernel: RTW: rtw_set_ps_mode(wlo1) Leave 802.11 power save - WIFI-TRAFFIC_BUSY
> Dec 25 22:56:36  kernel: RTW: rtl8723d_set_FwPwrMode_cmd(): FW LPS mode = 0, SmartPS=2
> Dec 25 22:56:38  kernel: RTW: rtw_set_ps_mode(wlo1) Enter 802.11 power save - WIFI-TRAFFIC_IDLE
> Dec 25 22:56:38  kernel: RTW: rtl8723d_set_FwPwrMode_cmd(): FW LPS mode = 2, SmartPS=2
> Dec 25 22:56:39  kernel: RTW: rtl8723d_fill_default_txdesc(wlo1): SP Packet(0x0806) rate=0x0
> Dec 25 22:56:40  kernel: RTW: OnAction_back
> Dec 25 22:56:40  kernel: RTW: OnAction_back, action=0
> Dec 25 22:56:40  kernel: RTW: issue_addba_rsp_wait_ack(wlo1) ra=98:de:d0:52:65:cc tid=1, acked, 1/3 in 2 ms
> Dec 25 22:56:40  kernel: RTW: traffic_status_watchdog(wlo1) acqiure wake_lock for 4500 ms(tx:16,rx_unicast:10)
> Dec 25 22:56:40  kernel: RTW: rtw_set_ps_mode(wlo1) Leave 802.11 power save - WIFI-TRAFFIC_BUSY
> Dec 25 22:56:40  kernel: RTW: rtl8723d_set_FwPwrMode_cmd(): FW LPS mode = 0, SmartPS=2
> Dec 25 22:56:41  kernel: RTW: rtw_issue_addbareq_cmd, p=0
> Dec 25 22:56:41  kernel: RTW: BA_starting_seqctrl = 223 for TID=0
> Dec 25 22:56:41  kernel: RTW: issue_addba_req(wlo1) ra=98:de:d0:52:65:cc tid=0
> Dec 25 22:56:41  kernel: RTW: OnAction_back
> Dec 25 22:56:41  kernel: RTW: OnAction_back, action=1
> Dec 25 22:56:41  kernel: RTW: agg_enable for TID=0
> Dec 25 22:56:42  kernel: RTW: traffic_status_watchdog(wlo1) acqiure wake_lock for 4500 ms(tx:101,rx_unicast:84)
> Dec 25 22:56:43  kernel: [UFW BLOCK] IN=wlo1 OUT= MAC=01:00:5e:00:00:01:98:de:d0:52:65:cc:08:00 SRC=192.168.0.1 DST=224.0.0.1 LEN=32 TOS=0x00 PREC=0xC0 TTL=1 ID=0 DF PROTO=2 
> Dec 25 22:56:44  kernel: RTW: traffic_status_watchdog(wlo1) acqiure wake_lock for 4500 ms(tx:23,rx_unicast:18)
> Dec 25 22:56:46  kernel: RTW: traffic_status_watchdog(wlo1) acqiure wake_lock for 4500 ms(tx:188,rx_unicast:159)
> Dec 25 22:56:48  kernel: RTW: traffic_status_watchdog(wlo1) acqiure wake_lock for 4500 ms(tx:42,rx_unicast:36)
> Dec 25 22:56:50  kernel: RTW: traffic_status_watchdog(wlo1) acqiure wake_lock for 4500 ms(tx:14,rx_unicast:10)
> Dec 25 22:56:52  kernel: RTW: traffic_status_watchdog(wlo1) acqiure wake_lock for 4500 ms(tx:47,rx_unicast:70)
> Dec 25 22:56:55  kernel: RTW: cfg80211_rtw_scan(wlo1)
> Dec 25 22:56:55  kernel: RTW: SetHwReg8723D(wlo1): [HW_VAR_MACID_SLEEP] macid=0, org reg_0x4d4=0x00000000
> Dec 25 22:56:55  kernel: RTW: ===>phy_SpurCalibration_8723D: Channel = 1
> Dec 25 22:56:55  kernel: RTW: ===>phy_SwChnl8723D: Channel = 1
> Dec 25 22:56:55  kernel: RTW: ===>phy_SpurCalibration_8723D: Channel = 2
> Dec 25 22:56:55  kernel: RTW: ===>phy_SwChnl8723D: Channel = 2
> Dec 25 22:56:55  kernel: RTW: ===>phy_SpurCalibration_8723D: Channel = 3
> Dec 25 22:56:55  kernel: RTW: ===>phy_SwChnl8723D: Channel = 3
> Dec 25 22:56:55  kernel: RTW: ===>phy_SpurCalibration_8723D: Channel = 4
> Dec 25 22:56:55  kernel: RTW: ===>phy_SwChnl8723D: Channel = 4
> Dec 25 22:56:55  kernel: RTW: ===>phy_SpurCalibration_8723D: Channel = 5
> Dec 25 22:56:55  kernel: RTW: ===>phy_SwChnl8723D: Channel = 5
> Dec 25 22:56:55  kernel: RTW: ===>phy_SpurCalibration_8723D: Channel = 6
> Dec 25 22:56:55  kernel: RTW: ===>phy_SwChnl8723D: Channel = 6
> Dec 25 22:56:55  kernel: RTW: ===>phy_SpurCalibration_8723D: Channel = 7
> Dec 25 22:56:55  kernel: RTW: ===>phy_SwChnl8723D: Channel = 7
> Dec 25 22:56:55  kernel: RTW: ===>phy_SpurCalibration_8723D: Channel = 8
> Dec 25 22:56:55  kernel: RTW: ===>phy_SwChnl8723D: Channel = 8
> Dec 25 22:56:55  kernel: RTW: ===>phy_SpurCalibration_8723D: Channel = 9
> Dec 25 22:56:55  kernel: RTW: ===>phy_SwChnl8723D: Channel = 9
> Dec 25 22:56:56  kernel: RTW: ===>phy_SpurCalibration_8723D: Channel = 10
> Dec 25 22:56:56  kernel: RTW: ===>phy_SwChnl8723D: Channel = 10
> Dec 25 22:56:56  kernel: RTW: ===>phy_SpurCalibration_8723D: Channel = 11
> Dec 25 22:56:56  kernel: RTW: ===>phy_SwChnl8723D: Channel = 11
> Dec 25 22:56:56  kernel: RTW: ===>phy_SpurCalibration_8723D: Channel = 12
> Dec 25 22:56:56  kernel: RTW: ===>phy_SwChnl8723D: Channel = 12
> Dec 25 22:56:56  kernel: RTW: ===>phy_SpurCalibration_8723D: Channel = 13
> Dec 25 22:56:56  kernel: RTW: ===>phy_SwChnl8723D: Channel = 13
> Dec 25 22:56:56  kernel: RTW: ===>phy_SpurCalibration_8723D: Channel = 4
> Dec 25 22:56:56  kernel: RTW: ===>phy_SwChnl8723D: Channel = 4
> Dec 25 22:56:56  kernel: RTW: SetHwReg8723D(wlo1): [HW_VAR_MACID_WAKEUP] macid=0, org reg_0x4d4=0x00000001
[... 8 lines cut ...]
> Dec 25 22:57:06  kernel: RTW: rtw_set_ps_mode(wlo1) Leave 802.11 power save - WIFI-TRAFFIC_BUSY
[... cut ...]
> Dec 25 22:57:31  kernel: RTW: rtw_set_ps_mode(wlo1) Enter 802.11 power save - WIFI-TRAFFIC_IDLE
> Dec 25 22:57:31  kernel: RTW: rtl8723d_set_FwPwrMode_cmd(): FW LPS mode = 2, SmartPS=2
> Dec 25 22:57:33  kernel: RTW: rtw_set_ps_mode(wlo1) Leave 802.11 power save - WIFI-TRAFFIC_BUSY
> Dec 25 22:57:33  kernel: RTW: rtl8723d_set_FwPwrMode_cmd(): FW LPS mode = 0, SmartPS=2
> Dec 25 22:57:35  kernel: RTW: rtw_set_ps_mode(wlo1) Enter 802.11 power save - WIFI-TRAFFIC_IDLE
> Dec 25 22:57:35  kernel: RTW: rtl8723d_set_FwPwrMode_cmd(): FW LPS mode = 2, SmartPS=2

jeremyb31, 2017-12-25:
Results from this command in terminal `grep [[:alnum:]] /sys/module/8723de/parameters/*`

jeremyb31, 2017-12-26:
Uninstall from dkms and try using mine at https://github.com/jeremyb31/rtl8723de I just made some changes in Makefile that might help, or you could look at https://github.com/jeremyb31/rtl8723de/commit/f2a316bd79385878df5fb95d888de74301f3b56d and do the changes manually on the source code you have and then reinstall

AvirahJ, 2018-01-16:
Hi @smlinux @jeremyb31 ,
I tried extracting the windows driver provided for rtl8723de bluetooth hoping to get some clues on the firmware. Since I'm not technically experienced, I cannot find anything reasonable that is related as firmware. The only file that matches "fw" (firmware) name is a .dll file called "rtl8723d_mp_chip_bt40_fw_asic_rom_patch_new.dll". This, I got after extracting two data.cab files in the archive. I have listed the contents of the extracted data1, data2 cab files part of WIndows driver here: https://pastebin.ca/3959861
This is the latest RTL Windows driver provided by HP : http://ftp.hp.com/pub/softpaq/sp83501-84000/sp83944.exe

I was thinking, if it is possible to get a firmware dump of bluetooth from a Windows 10 laptop with RTL8723DE OR extract the bluetooth firmware from Windows drivers. 
Is there any chances, if this will work?

AvirahJ, 2018-03-06:
The firmware for bluetooth is listed as available now in kernel firmware git tree:
https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/commit/?id=6d5131107f2ba67a13f469ac770a55f101ba654d

I have git cloned and copied the contents of rtl_usb which includes rtl8723d firmware and config bin to /lib/firmware/rtl_bt and tried depmod -a. There is no effect and still the hardware is telling about 8723b bt firmware.

:~$ dmesg |grep -i bluetooth

[   10.784665] Bluetooth: Core ver 2.22
[   10.784679] Bluetooth: HCI device and connection manager initialized
[   10.784682] Bluetooth: HCI socket layer initialized
[   10.784683] Bluetooth: L2CAP socket layer initialized
[   10.784688] Bluetooth: SCO socket layer initialized
[   10.863701] Bluetooth: hci0: rtl: examining hci_ver=08 hci_rev=000d lmp_ver=08 lmp_subver=8723
[   10.863702] Bluetooth: hci0: rtl: loading rtl_bt/rtl8723b_config.bin
[   10.971044] bluetooth hci0: Direct firmware load for rtl_bt/rtl8723b_config.bin failed with error -2
[   10.971047] Bluetooth: hci0: rtl: loading rtl_bt/rtl8723b_fw.bin
[   10.995696] Bluetooth: hci0: rom_version status=0 version=2
[   10.995698] Bluetooth: hci0: didn't find patch for chip id 2
[   21.767200] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   21.767202] Bluetooth: BNEP filters: protocol multicast
[   21.767206] Bluetooth: BNEP socket layer initialized

AvirahJ, 2018-03-06:
The firmware for bluetooth is listed as available now in kernel firmware git tree:
https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/commit/?id=6d5131107f2ba67a13f469ac770a55f101ba654d

I have git cloned and copied the contents of rtl_usb which includes rtl8723d firmware and config bin to /lib/firmware/rtl_bt and tried depmod -a. There is no effect and still the hardware is telling about 8723b bt firmware.

:~$ dmesg |grep -i bluetooth

[   10.784665] Bluetooth: Core ver 2.22
[   10.784679] Bluetooth: HCI device and connection manager initialized
[   10.784682] Bluetooth: HCI socket layer initialized
[   10.784683] Bluetooth: L2CAP socket layer initialized
[   10.784688] Bluetooth: SCO socket layer initialized
[   10.863701] Bluetooth: hci0: rtl: examining hci_ver=08 hci_rev=000d lmp_ver=08 lmp_subver=8723
[   10.863702] Bluetooth: hci0: rtl: loading rtl_bt/rtl8723b_config.bin
[   10.971044] bluetooth hci0: Direct firmware load for rtl_bt/rtl8723b_config.bin failed with error -2
[   10.971047] Bluetooth: hci0: rtl: loading rtl_bt/rtl8723b_fw.bin
[   10.995696] Bluetooth: hci0: rom_version status=0 version=2
[   10.995698] Bluetooth: hci0: didn't find patch for chip id 2
[   21.767200] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   21.767202] Bluetooth: BNEP filters: protocol multicast
[   21.767206] Bluetooth: BNEP socket layer initialized

So, 8723B firmware is getting detected instead of 8723D.

agsbenitez, 2018-03-21:
@AvirahJ Can you detail the procedure?

alphacentauri82, 2018-04-27:
Guys, try this, and make sure you checkout the extended branch to make:
https://github.com/lwfinger/rtlwifi_new/tree/extended

cburschka, 2018-05-17:
The bluetooth still doesn't work for me, but according to https://github.com/lwfinger/rtlwifi_new/issues/353, this will finally be rectified by the yet unreleased 4.17 kernel (which I suspect will become stable some time in the next two months).

AvirahJ, 2018-05-18:
Yes, Bluetooth is working good with 4.17-rc kernel. Kernel versions prior to 4.17 refers the rtl8723b firmware (which is wrong firmware) and ends up with the error "hci0: didn't find patch for chip id 2".

kerozeno, 2018-05-21:
@AvirahJ it works for me, i can use bluetooth or wifi now, but, when i am connected to wifi network, bluetooth connection is lost. only way that i have to view visibles bluetooth devices and connect them is if wifi is not connected

vivek-aries, 2018-05-30:
@AvirahJ I was also having the same problem. But I got help from here. I am using kernel 4.16.13 on Ubuntu so it works even without 4.17.x

You have to add this repository:

sudo add-apt-repository ppa:hanipouspilot/bluetooth
sudo apt update

and then install this package:

sudo apt install btrtl-rtl8723de-dkms

It is available here: 
https://launchpad.net/~hanipouspilot/+archive/ubuntu/bluetooth

konbeto, 2018-06-08:
@viveikjha Thank you. That got my BT working on Ubuntu 18.04 using kernel **4.15.0**.

glodfinch, 2018-06-16:
@konbeto are you able to stream audio to bluetooth headphones? I am also on Ubuntu 18.04 with 4.15.0, and while I can detect and connect to my headset after following @viveikjha's instructions, it will not select them in the sound menu in settings.

glodfinch, 2018-09-24:
I'd like to add that I probably jumped the gun with bluetooth working on later kernels, as has become clear by now. I'm currently running 4.18.9. Since 4.17 bluetooth has worked purely from the rtlwifi_new repo by lwfinger without additional setup. However the issue seems to be it has a poor signal quality. My bluetooth mouse works ok as long as it's within roughly a meter of my laptop, but bluetooth speakers of any kind will almost immediately start dropping audio and will usually disconnect completely after a little while. As far as I know there's no solution to this at the moment.

padlych, 2019-01-30:
Linux kernel 4.17.3-041703-generic. The same situation. Bluetooth began to work with
[ 10.647823] Bluetooth: hci0: rtl: loading rtl_bt/rtl8723d_config.bin
[ 10.820288] Bluetooth: hci0: rtl: loading rtl_bt/rtl8723d_fw.bin
But link quality is poor (about 10%). So headset do not work with A2DP profile (sound is buggy because bandwidth is poor)