smlinux/rtl8723de #5: [dkms error] ERROR (dkms apport): binary package for rtl8723de: 5.1.1.8_21285.20171026_COEX20170111-1414 not found

AvirahJ, 2017-12-23:
I tried running:
```
:~# dkms add ./rtl8723de
:~# dkms build rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
make -j4 KERNELRELEASE=4.14.0-8.1-liquorix-amd64 -C /lib/modules/4.14.0-8.1-liquorix-amd64/build M=/var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build...
ERROR (dkms apport): binary package for rtl8723de: 5.1.1.8_21285.20171026_COEX20170111-1414 not found
Error!  Build of 8723de.ko failed for: 4.14.0-8.1-liquorix-amd64 (x86_64)
Consult the make.log in the build directory
/var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/ for more information.

```

The make.log says no module is being built:
`:/var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414$ cat build/make.log 
DKMS make.log for rtl8723de-5.1.1.8_21285.20171026_COEX20170111-1414 for kernel 4.14.0-8.1-liquorix-amd64 (x86_64)
Sat Dec 23 09:47:35 IST 2017
make: Entering directory '/usr/src/linux-headers-4.14.0-8.1-liquorix-amd64'
  AR      /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/built-in.o
  Building modules, stage 2.
  MODPOST 0 modules
make: Leaving directory '/usr/src/linux-headers-4.14.0-8.1-liquorix-amd64'`

Unfortunately, I have no programming knowledge. :( . But, saw a similar error which says some variables are not mentioned in the makefile?

https://github.com/lwfinger/rtl8723au/issues/10

AvirahJ, 2017-12-23:
Please note that, make and sudo make install compiles fine.

castellanos94, 2017-12-23:
Same problem, any idea to fix that ?

jeremyb31, 2017-12-23:
I might have the fix on my fork https://github.com/jeremyb31/rtl8723de
I made a small change to the dkms.conf file and it works
https://github.com/jeremyb31/rtl8723de/commit/9eac941bfec813b4419e61df71f1f8e9fe2558f2

castellanos94, 2017-12-23:
Thanks @jeremyb31 , thats works for me.

smlinux, 2017-12-23:
Thanks **jeremyb31**, fixed.

closed 2017-12-23 by smlinux

sergehijo, 2018-01-18:
Thanks @jeremyb31 !! Been waiting to install the driver correctly for almost six months, then some problems with the kernel versions lol a nightmare.  Thanks again!

Carsano, 2018-07-25:
With your repo @jeremyb31 I have a problem : 
Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
'make' all KVER=4.15.0-29-generic...(bad exit status: 2)
ERROR (dkms apport): binary package for rtl8723de: 5.1.1.8_21285.20171026_COEX20170111-1414 not found
Error! Bad return status for module build on kernel: 4.15.0-29-generic (x86_64)
Consult /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/make.log for more information.
Anyone has an answer ?

jeremyb31, 2018-07-26:
I would use Larry Fingers rtlwifi_new extended branch in 4.15
https://github.com/lwfinger/rtlwifi_new/tree/extended

kol4k, 2018-07-31:
Yess i have same issue like @Carsano

nishantkohli94, 2018-08-06:
@jeremyb31 @smlinux 

 bro just 1 month before  i cloned this git repository and added the driver  for ubuntu 16 it was going easy.

please check what changes has taken place . i am unable to install driver.

and getting below error.

ERROR (dkms apport): binary package for rtl8723de: 5.1.1.8_21285.20171026_COEX20170111-1414 not found

waiting for your kind help.

jeremyb31, 2018-08-06:
I would suggest using Larry Fingers rtlwifi_new. To install see https://gist.github.com/jeremyb31/8687eec6ea7f9dca9e71a24aab197f0e
After rebooting if you notice low signal strength, try
`sudo modprobe -r rtl8723de && sleep 10 && sudo modprobe rtl8723de ant_sel=2`

nishantkohli94, 2018-08-11:
[image]

after installing the driver from above given link i am getting "wifi Network not ready".

jeremyb31, 2018-08-11:
What is the results for 
`dmesg | grep rtl`

nishantkohli94, 2018-08-12:
@jeremyb31 

dmesg | grep rtl
[   20.213442] Bluetooth: hci0: rtl: examining hci_ver=08 hci_rev=000d lmp_ver=08 lmp_subver=8723
[   20.213443] Bluetooth: hci0: rtl: loading rtl_bt/rtl8723b_config.bin
[   20.221864] rtlwifi: loading out-of-tree module taints kernel.
[   20.221940] rtlwifi: module verification failed: signature and/or required key missing - tainting kernel
[   20.320371] bluetooth hci0: Direct firmware load for rtl_bt/rtl8723b_config.bin failed with error -2
[   20.320373] Bluetooth: hci0: rtl: loading rtl_bt/rtl8723b_fw.bin
[   20.407431] Bluetooth: hci0: rtl: examining hci_ver=08 hci_rev=000d lmp_ver=08 lmp_subver=8723
[   20.407432] Bluetooth: hci0: rtl: loading rtl_bt/rtl8723b_config.bin
[   20.407442] bluetooth hci0: Direct firmware load for rtl_bt/rtl8723b_config.bin failed with error -2
[   20.407443] Bluetooth: hci0: rtl: loading rtl_bt/rtl8723b_fw.bin
[   20.549271] rtl8723de: Using firmware rtlwifi/rtl8723defw.bin
[   20.551612] ieee80211 phy0: Selected rate control algorithm 'rtl_rc'
[   20.551770] rtlwifi: rtlwifi: wireless switch is on
[   20.552544] rtl8723de 0000:03:00.0 wlo1: renamed from wlan0
[   20.588256] rtl8723de 0000:03:00.0: Direct firmware load for rtlwifi/rtl8723defw.bin failed with error -2
[   20.588258] rtlwifi: Selected firmware is not available
[34488.065876] rtlwifi: rtlwifi: wireless switch is on
[34489.235898] Bluetooth: hci0: rtl: examining hci_ver=08 hci_rev=000d lmp_ver=08 lmp_subver=8723
[34489.235901] Bluetooth: hci0: rtl: loading rtl_bt/rtl8723b_config.bin
[34489.235917] bluetooth hci0: Direct firmware load for rtl_bt/rtl8723b_config.bin failed with error -2
[34489.235919] Bluetooth: hci0: rtl: loading rtl_bt/rtl8723b_fw.bin
[34489.245811] Bluetooth: hci0: rtl: examining hci_ver=08 hci_rev=000d lmp_ver=08 lmp_subver=8723
[34489.245812] Bluetooth: hci0: rtl: loading rtl_bt/rtl8723b_config.bin
[34489.245827] bluetooth hci0: Direct firmware load for rtl_bt/rtl8723b_config.bin failed with error -2
[34489.245829] Bluetooth: hci0: rtl: loading rtl_bt/rtl8723b_fw.bin
[34556.165791] rtlwifi: rtlwifi: wireless switch is on
[34557.047785] Bluetooth: hci0: rtl: examining hci_ver=08 hci_rev=000d lmp_ver=08 lmp_subver=8723
[34557.047787] Bluetooth: hci0: rtl: loading rtl_bt/rtl8723b_config.bin
[34557.047802] bluetooth hci0: Direct firmware load for rtl_bt/rtl8723b_config.bin failed with error -2
[34557.047803] Bluetooth: hci0: rtl: loading rtl_bt/rtl8723b_fw.bin
[34557.070780] Bluetooth: hci0: rtl: examining hci_ver=08 hci_rev=000d lmp_ver=08 lmp_subver=8723
[34557.070783] Bluetooth: hci0: rtl: loading rtl_bt/rtl8723b_config.bin
[34557.070798] bluetooth hci0: Direct firmware load for rtl_bt/rtl8723b_config.bin failed with error -2
[34557.070800] Bluetooth: hci0: rtl: loading rtl_bt/rtl8723b_fw.bin

jeremyb31, 2018-08-13:
In terminal
`cd /lib/firmware/rtlwifi`
`sudo wget https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/plain/rtlwifi/rtl8723defw.bin`
Reboot

nishantkohli94, 2018-08-14:
@jeremyb31  thank you brother.

Its working absolutely fine now.

thanks a lot . what a pleasant start of the day you have given. 

:100:

mmairs, 2019-03-04:
@jeremyb31 thx*1000000 to you and Mr. Fingers!

sorted.

felipecolombo, 2019-05-16:
This https://github.com/cilynx/rtl88x2BU_WiFi_linux_v5.3.1_27678.20180430_COEX20180427-5959

worked for my Sonkir USB 3.0 WiFi Adapter AC 1200Mbps 5G/2.4G Dual Band RTL8811/RTL8812 under Ubuntu 18.10 4.18.0-17-generic

kh0kv, 2019-12-01:
> I tried running:
> 
> ```
> :~# dkms add ./rtl8723de
> :~# dkms build rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414
> 
> Kernel preparation unnecessary for this kernel.  Skipping...
> 
> Building module:
> cleaning build area...
> make -j4 KERNELRELEASE=4.14.0-8.1-liquorix-amd64 -C /lib/modules/4.14.0-8.1-liquorix-amd64/build M=/var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build...
> ERROR (dkms apport): binary package for rtl8723de: 5.1.1.8_21285.20171026_COEX20170111-1414 not found
> Error!  Build of 8723de.ko failed for: 4.14.0-8.1-liquorix-amd64 (x86_64)
> Consult the make.log in the build directory
> /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/ for more information.
> ```
> 
> The make.log says no module is being built:
> `:/var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414$ cat build/make.log DKMS make.log for rtl8723de-5.1.1.8_21285.20171026_COEX20170111-1414 for kernel 4.14.0-8.1-liquorix-amd64 (x86_64) Sat Dec 23 09:47:35 IST 2017 make: Entering directory '/usr/src/linux-headers-4.14.0-8.1-liquorix-amd64' AR /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/built-in.o Building modules, stage 2. MODPOST 0 modules make: Leaving directory '/usr/src/linux-headers-4.14.0-8.1-liquorix-amd64'`
> 
> Unfortunately, I have no programming knowledge. :( . But, saw a similar error which says some variables are not mentioned in the makefile?
> 
> [lwfinger/rtl8723au#10](https://github.com/lwfinger/rtl8723au/issues/10)

japrog, 2020-04-07:
Hi,

Thanks a lot for your work. It's work fine for my Xubuntu 18.04 4.15.0.91-generic kernel.
I've been followed the post https://github.com/smlinux/rtl8723de/issues/5#issuecomment-410850664.

JafGit, 2020-06-12:
Thank you @jeremyb31 It was a great help after so much of struggle to find the correct solution.
Much appreciated.

soldadurabog, 2020-12-10:
Try this one, it worked for me https://github.com/vvmspace/rtl8723de I had the exact same problem, the package seems to have problem with 4.15 kernel