smlinux/rtl8723de #6: issues while installing on HP -15Q BU007TU

sanketdhole, 2017-12-23:
I got some errors while installing the package on my PC with running Ubuntu 16.04 LTS. ]

> sanku@MyPC:~/rtl8723de$ make
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/4.10.0-28-generic/build M=/home/sanku/rtl8723de  modules
make[1]: Entering directory '/usr/src/linux-headers-4.10.0-28-generic'
  CC [M]  /home/sanku/rtl8723de/core/rtw_cmd.o
In file included from /home/sanku/rtl8723de/include/drv_types.h:32:0,
                 from /home/sanku/rtl8723de/core/rtw_cmd.c:22:
/home/sanku/rtl8723de/include/osdep_service.h:41:33: fatal error: linux/sched/signal.h: No such file or directory
compilation terminated.
scripts/Makefile.build:294: recipe for target '/home/sanku/rtl8723de/core/rtw_cmd.o' failed
make[2]: *** [/home/sanku/rtl8723de/core/rtw_cmd.o] Error 1
Makefile:1524: recipe for target '_module_/home/sanku/rtl8723de' failed
make[1]: *** [_module_/home/sanku/rtl8723de] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-4.10.0-28-generic'
Makefile:1884: recipe for target 'modules' failed
make: *** [modules] Error 2

smlinux, 2017-12-23:
Kernel 4.10 - use 4.10-down branch.

closed 2017-12-23 by smlinux

sanketdhole, 2017-12-24:
Thanks man. That solved the issue....