smlinux/rtl8723de #8: Installation error

Doxdrum, 2017-12-27:
I'm following the instruction to install the module, and got the following error. 

```
KMS make.log for rtl8723de-5.1.1.8_21285.20171026_COEX20170111-1414 for kernel 4.14.0-2-amd64 (x86_64)
Wed Dec 27 16:16:48 -03 2017
make: Entering directory '/usr/src/linux-headers-4.14.0-2-amd64'
  AR      /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/built-in.o
  Building modules, stage 2.
  MODPOST 0 modules
make: Leaving directory '/usr/src/linux-headers-4.14.0-2-amd64'
```
What is the problem? The kernel???

ggemmill, 2017-12-28:
Similar problem on Mageia6-x64:
dkms add ./rtl8723de

Error! Invalid number of arguments passed.
Usage: add -m <module> -v <module-version>
How do I find the module-version?
Thanks
Grimble

zolleride, 2018-01-06:
To install the module for the RTL8723DE chipset, first determine your kernel using uname -r if your kernel is lower than 4.11, do

sudo apt-get install build-essential dkms git
git clone -b 4.10-down https://github.com/jeremyb31/rtl8723de.git
sudo dkms add ./rtl8723de
sudo dkms install rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414
Then reboot
If you have kernel 4.11 and newer do

sudo apt-get install build-essential dkms git
git clone https://github.com/jeremyb31/rtl8723de.git
sudo dkms add ./rtl8723de
sudo dkms install rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414
Then reboot

haffla, 2018-01-12:
@zolleride thank you. The [jeremyb31](https://github.com/jeremyb31/rtl8723de) fork works for me.

gravitydestroyer, 2018-02-08:
My kernel version is higher than 4.11(4.4.0-31-generic).When i am trying to install the module,i get the following error message:
`Building module:
cleaning build area....
'make' all KVER=4.4.0-31-generic....(bad exit status: 2)
ERROR (dkms apport): binary package for rtl8723de: 5.1.1.8_21285.20171026_COEX20170111-1414 not found
Error! Bad return status for module build on kernel: 4.4.0-31-generic (x86_64)
Consult /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/make.log for more information.`

haffla, 2018-02-08:
@gravitydestroyer `4.4` is not higher than `4.11`, 11 > 4!

lenny87, 2018-02-08:
@gravitydestroyer install HWE kernel on your ubuntu and compile again.
https://wiki.ubuntu.com/Kernel/LTSEnablementStack#Desktop

RituparnaLinux, 2018-03-07:
@zolleride Your solution worked for me. My kernel is 4.13.0-36, the 2nd option clicked!

I have been struggling with this problem for 3 days. Thank you.

shakier21, 2019-12-25:
HI, can someone please assist me, my wifi is not working. i am using 5.0.0.37 kernel and i am getting an error when running the commands

sudo dkms install rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414 

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
'make' all KVER=5.0.0-37-generic...(bad exit status: 2)
Error! Bad return status for module build on kernel: 5.0.0-37-generic (x86_64)
Consult /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/make.log for more information.

ManojVarma101, 2020-11-05:
> To install the module for the RTL8723DE chipset, first determine your kernel using uname -r if your kernel is lower than 4.11, do
>
> sudo apt-get install build-essential dkms git
> git clone -b [...]

My kernel version is 4.15.0-122-generic

Even after following above steps for version > 4.11 the same error is arising

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
'make' all KVER=4.15.0-122-generic.....(bad exit status: 2)
ERROR (dkms apport): binary package for rtl8723de: 5.1.1.8_21285.20171026_COEX20170111-1414 not found
Error! Bad return status for module build on kernel: 4.15.0-122-generic (x86_64)
Consult /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/make.log for more information.