smlinux/rtl8723de #9: Installation error

haffla, 2018-01-03:
I've no clue. Can only paste the log here

```
DKMS make.log for rtl8723de-5.1.1.8_21285.20171026_COEX20170111-1414 for kernel 4.11.0-14-generic (x86_64)
Mi 3. Jan 16:04:22 CET 2018
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/4.11.0-14-generic/build M=/var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build  modules
make[1]: Verzeichnis „/usr/src/linux-headers-4.11.0-14-generic“ wird betreten
  CC [M]  /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/core/rtw_cmd.o
  CC [M]  /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/core/rtw_security.o
  CC [M]  /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/core/rtw_debug.o
  CC [M]  /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/core/rtw_io.o
  CC [M]  /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/core/rtw_ioctl_query.o
  CC [M]  /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/core/rtw_ioctl_set.o
  CC [M]  /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/core/rtw_ieee80211.o
  CC [M]  /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/core/rtw_mlme.o
  CC [M]  /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/core/rtw_mlme_ext.o
  CC [M]  /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/core/rtw_mi.o
  CC [M]  /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/core/rtw_wlan_util.o
  CC [M]  /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/core/rtw_vht.o
  CC [M]  /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/core/rtw_pwrctrl.o
  CC [M]  /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/core/rtw_rf.o
  CC [M]  /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/core/rtw_recv.o
  CC [M]  /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/core/rtw_sta_mgt.o
  CC [M]  /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/core/rtw_ap.o
  CC [M]  /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/core/rtw_xmit.o
  CC [M]  /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/core/rtw_p2p.o
  CC [M]  /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/core/rtw_tdls.o
  CC [M]  /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/core/rtw_br_ext.o
  CC [M]  /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/core/rtw_iol.o
  CC [M]  /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/core/rtw_sreset.o
  CC [M]  /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/core/rtw_btcoex.o
  CC [M]  /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/core/rtw_beamforming.o
  CC [M]  /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/core/rtw_odm.o
  CC [M]  /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/core/efuse/rtw_efuse.o
  CC [M]  /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/os_dep/osdep_service.o
  CC [M]  /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/os_dep/linux/os_intfs.o
  CC [M]  /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/os_dep/linux/pci_intf.o
  CC [M]  /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/os_dep/linux/pci_ops_linux.o
  CC [M]  /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/os_dep/linux/ioctl_linux.o
  CC [M]  /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/os_dep/linux/xmit_linux.o
  CC [M]  /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/os_dep/linux/mlme_linux.o
  CC [M]  /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/os_dep/linux/recv_linux.o
  CC [M]  /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/os_dep/linux/ioctl_cfg80211.o
/var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/os_dep/linux/ioctl_cfg80211.c: In function ‘rtw_cfg80211_indicate_connect’:
/var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/os_dep/linux/ioctl_cfg80211.c:744:10: error: variable ‘roam_info’ has initializer but incomplete type
   struct cfg80211_roam_info roam_info = {};
          ^
/var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/os_dep/linux/ioctl_cfg80211.c:744:29: error: storage size of ‘roam_info’ isn’t known
   struct cfg80211_roam_info roam_info = {};
                             ^
In file included from ./include/linux/kmod.h:22:0,
                 from ./include/linux/module.h:13,
                 from /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/include/basic_types.h:81,
                 from /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/include/drv_types.h:31,
                 from /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/os_dep/linux/ioctl_cfg80211.c:22:
./include/linux/gfp.h:239:20: warning: passing argument 3 of ‘cfg80211_roamed’ makes pointer from integer without a cast [-Wint-conversion]
 #define GFP_ATOMIC (__GFP_HIGH|__GFP_ATOMIC|__GFP_KSWAPD_RECLAIM)
                    ^
/var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/os_dep/linux/ioctl_cfg80211.c:760:50: note: in expansion of macro ‘GFP_ATOMIC’
   cfg80211_roamed(padapter->pnetdev, &roam_info, GFP_ATOMIC);
                                                  ^
In file included from /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/include/osdep_service_linux.h:91:0,
                 from /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/include/osdep_service.h:42,
                 from /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/include/drv_types.h:32,
                 from /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/os_dep/linux/ioctl_cfg80211.c:22:
./include/net/cfg80211.h:5233:6: note: expected ‘const u8 * {aka const unsigned char *}’ but argument is of type ‘unsigned int’
 void cfg80211_roamed(struct net_device *dev,
      ^
/var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/os_dep/linux/ioctl_cfg80211.c:760:3: error: too few arguments to function ‘cfg80211_roamed’
   cfg80211_roamed(padapter->pnetdev, &roam_info, GFP_ATOMIC);
[... 8 lines cut ...]
/var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/os_dep/linux/ioctl_cfg80211.c: In function ‘rtw_cfg80211_preinit_wiphy’:
/var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/os_dep/linux/ioctl_cfg80211.c:6762:7: error: ‘struct wiphy’ has no member named ‘max_sched_scan_reqs’
  wiphy->max_sched_scan_reqs = 1;
       ^
/var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/os_dep/linux/ioctl_cfg80211.c: At top level:
/var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/os_dep/linux/ioctl_cfg80211.c:6795:25: error: initialization from incompatible pointer type [-Werror=incompatible-pointer-types]
  .change_virtual_intf = cfg80211_rtw_change_iface,
                         ^
/var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/os_dep/linux/ioctl_cfg80211.c:6795:25: note: (near initialization for ‘rtw_cfg80211_ops.change_virtual_intf’)
/var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/os_dep/linux/ioctl_cfg80211.c:6818:22: error: initialization from incompatible pointer type [-Werror=incompatible-pointer-types]
  .add_virtual_intf = cfg80211_rtw_add_virtual_intf,
                      ^
/var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/os_dep/linux/ioctl_cfg80211.c:6818:22: note: (near initialization for ‘rtw_cfg80211_ops.add_virtual_intf’)
cc1: some warnings being treated as errors
scripts/Makefile.build:294: die Regel für Ziel „/var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/os_dep/linux/ioctl_cfg80211.o“ scheiterte
make[2]: *** [/var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/os_dep/linux/ioctl_cfg80211.o] Fehler 1
Makefile:1526: die Regel für Ziel „_module_/var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build“ scheiterte
make[1]: *** [_module_/var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build] Fehler 2
make[1]: Verzeichnis „/usr/src/linux-headers-4.11.0-14-generic“ wird verlassen
Makefile:1884: die Regel für Ziel „modules“ scheiterte
make: *** [modules] Fehler 2

```

Any idea?

smlinux, 2018-01-06:
What Linux distro use?

haffla, 2018-01-06:
That's Ubuntu 16.04. Kernel updated to 4.11.

bughuntercat, 2018-01-08:
I have the same problem. Debian buster, kernel 4.14

haffla, 2018-01-12:
https://github.com/jeremyb31/rtl8723de.git

This fork is working for me :muscle:

jaseemharry, 2018-06-04:
I cant still get it to work. Please help. 
cd rtl8723de
make
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/4.11.12-041112-generic/build M=/home/rehmathulla/rtl8723de  modules
make[1]: Entering directory `/usr/src/linux-headers-4.11.12-041112-generic'
  CC [M]  /home/rehmathulla/rtl8723de/core/rtw_cmd.o
  CC [M]  /home/rehmathulla/rtl8723de/core/rtw_security.o
  CC [M]  /home/rehmathulla/rtl8723de/core/rtw_debug.o
  CC [M]  /home/rehmathulla/rtl8723de/core/rtw_io.o
  CC [M]  /home/rehmathulla/rtl8723de/core/rtw_ioctl_query.o
  CC [M]  /home/rehmathulla/rtl8723de/core/rtw_ioctl_set.o
  CC [M]  /home/rehmathulla/rtl8723de/core/rtw_ieee80211.o
  CC [M]  /home/rehmathulla/rtl8723de/core/rtw_mlme.o
  CC [M]  /home/rehmathulla/rtl8723de/core/rtw_mlme_ext.o
  CC [M]  /home/rehmathulla/rtl8723de/core/rtw_mi.o
  CC [M]  /home/rehmathulla/rtl8723de/core/rtw_wlan_util.o
  CC [M]  /home/rehmathulla/rtl8723de/core/rtw_vht.o
  CC [M]  /home/rehmathulla/rtl8723de/core/rtw_pwrctrl.o
  CC [M]  /home/rehmathulla/rtl8723de/core/rtw_rf.o
  CC [M]  /home/rehmathulla/rtl8723de/core/rtw_recv.o
  CC [M]  /home/rehmathulla/rtl8723de/core/rtw_sta_mgt.o
  CC [M]  /home/rehmathulla/rtl8723de/core/rtw_ap.o
  CC [M]  /home/rehmathulla/rtl8723de/core/rtw_xmit.o
  CC [M]  /home/rehmathulla/rtl8723de/core/rtw_p2p.o
  CC [M]  /home/rehmathulla/rtl8723de/core/rtw_tdls.o
  CC [M]  /home/rehmathulla/rtl8723de/core/rtw_br_ext.o
  CC [M]  /home/rehmathulla/rtl8723de/core/rtw_iol.o
  CC [M]  /home/rehmathulla/rtl8723de/core/rtw_sreset.o
  CC [M]  /home/rehmathulla/rtl8723de/core/rtw_btcoex.o
  CC [M]  /home/rehmathulla/rtl8723de/core/rtw_beamforming.o
  CC [M]  /home/rehmathulla/rtl8723de/core/rtw_odm.o
  CC [M]  /home/rehmathulla/rtl8723de/core/efuse/rtw_efuse.o
  CC [M]  /home/rehmathulla/rtl8723de/os_dep/osdep_service.o
  CC [M]  /home/rehmathulla/rtl8723de/os_dep/linux/os_intfs.o
  CC [M]  /home/rehmathulla/rtl8723de/os_dep/linux/pci_intf.o
  CC [M]  /home/rehmathulla/rtl8723de/os_dep/linux/pci_ops_linux.o
  CC [M]  /home/rehmathulla/rtl8723de/os_dep/linux/ioctl_linux.o
  CC [M]  /home/rehmathulla/rtl8723de/os_dep/linux/xmit_linux.o
  CC [M]  /home/rehmathulla/rtl8723de/os_dep/linux/mlme_linux.o
  CC [M]  /home/rehmathulla/rtl8723de/os_dep/linux/recv_linux.o
  CC [M]  /home/rehmathulla/rtl8723de/os_dep/linux/ioctl_cfg80211.o
/home/rehmathulla/rtl8723de/os_dep/linux/ioctl_cfg80211.c: In function ‘rtw_cfg80211_indicate_connect’:
/home/rehmathulla/rtl8723de/os_dep/linux/ioctl_cfg80211.c:744:10: error: variable ‘roam_info’ has initializer but incomplete type
   struct cfg80211_roam_info roam_info = {};
          ^
/home/rehmathulla/rtl8723de/os_dep/linux/ioctl_cfg80211.c:744:29: error: storage size of ‘roam_info’ isn’t known
   struct cfg80211_roam_info roam_info = {};
                             ^
In file included from ./include/linux/kmod.h:22:0,
                 from ./include/linux/module.h:13,
                 from /home/rehmathulla/rtl8723de/include/basic_types.h:81,
                 from /home/rehmathulla/rtl8723de/include/drv_types.h:31,
                 from /home/rehmathulla/rtl8723de/os_dep/linux/ioctl_cfg80211.c:22:
./include/linux/gfp.h:239:20: warning: passing argument 3 of ‘cfg80211_roamed’ makes pointer from integer without a cast [-Wint-conversion]
 #define GFP_ATOMIC (__GFP_HIGH|__GFP_ATOMIC|__GFP_KSWAPD_RECLAIM)
                    ^
/home/rehmathulla/rtl8723de/os_dep/linux/ioctl_cfg80211.c:760:50: note: in expansion of macro ‘GFP_ATOMIC’
   cfg80211_roamed(padapter->pnetdev, &roam_info, GFP_ATOMIC);
                                                  ^
In file included from /home/rehmathulla/rtl8723de/include/osdep_service_linux.h:91:0,
                 from /home/rehmathulla/rtl8723de/include/osdep_service.h:42,
                 from /home/rehmathulla/rtl8723de/include/drv_types.h:32,
                 from /home/rehmathulla/rtl8723de/os_dep/linux/ioctl_cfg80211.c:22:
./include/net/cfg80211.h:5233:6: note: expected ‘const u8 * {aka const unsigned char *}’ but argument is of type ‘unsigned int’
 void cfg80211_roamed(struct net_device *dev,
      ^
/home/rehmathulla/rtl8723de/os_dep/linux/ioctl_cfg80211.c:760:3: error: too few arguments to function ‘cfg80211_roamed’
   cfg80211_roamed(padapter->pnetdev, &roam_info, GFP_ATOMIC);
   ^
In file included from /home/rehmathulla/rtl8723de/include/osdep_service_linux.h:91:0,
[... 1 lines cut ...]
                 from /home/rehmathulla/rtl8723de/include/drv_types.h:32,
                 from /home/rehmathulla/rtl8723de/os_dep/linux/ioctl_cfg80211.c:22:
./include/net/cfg80211.h:5233:6: note: declared here
 void cfg80211_roamed(struct net_device *dev,
      ^
/home/rehmathulla/rtl8723de/os_dep/linux/ioctl_cfg80211.c: In function ‘rtw_cfg80211_preinit_wiphy’:
/home/rehmathulla/rtl8723de/os_dep/linux/ioctl_cfg80211.c:6762:7: error: ‘struct wiphy’ has no member named ‘max_sched_scan_reqs’
  wiphy->max_sched_scan_reqs = 1;
       ^
/home/rehmathulla/rtl8723de/os_dep/linux/ioctl_cfg80211.c: At top level:
/home/rehmathulla/rtl8723de/os_dep/linux/ioctl_cfg80211.c:6795:25: error: initialization from incompatible pointer type [-Werror=incompatible-pointer-types]
  .change_virtual_intf = cfg80211_rtw_change_iface,
                         ^
/home/rehmathulla/rtl8723de/os_dep/linux/ioctl_cfg80211.c:6795:25: note: (near initialization for ‘rtw_cfg80211_ops.change_virtual_intf’)
[... cut ...]
  .add_virtual_intf = cfg80211_rtw_add_virtual_intf,
                      ^
/home/rehmathulla/rtl8723de/os_dep/linux/ioctl_cfg80211.c:6818:22: note: (near initialization for ‘rtw_cfg80211_ops.add_virtual_intf’)
cc1: some warnings being treated as errors
make[2]: *** [/home/rehmathulla/rtl8723de/os_dep/linux/ioctl_cfg80211.o] Error 1
make[1]: *** [_module_/home/rehmathulla/rtl8723de] Error 2
make[1]: Leaving directory `/usr/src/linux-headers-4.11.12-041112-generic'
make: *** [modules] Error 2

Somebody please help me.