smlinux/rtl8723de #13: Problems when compiling in Linux 4.15.2-2

c4explosive, 2018-02-13:
The driver throws a error when compiling with Linux 4.15 in Archlinux. The error is about "data member" of struct timer_list. The cause is probably the change of timer interface in Linux kernel [(https://lkml.org/lkml/2017/11/25/90)](https://lkml.org/lkml/2017/11/25/90)

Compile's output error is detailed here:

make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/4.15.2-2-ARCH/build M=/ArchBootstraping271217/rtl8723de-4.11-up  modules
make[1]: Entering directory '/usr/lib/modules/4.15.2-2-ARCH/build'
  CC [M]  /ArchBootstraping271217/rtl8723de-4.11-up/core/rtw_cmd.o
In file included from /ArchBootstraping271217/rtl8723de-4.11-up/include/osdep_service.h:42:0,
                 from /ArchBootstraping271217/rtl8723de-4.11-up/include/drv_types.h:32,
                 from /ArchBootstraping271217/rtl8723de-4.11-up/core/rtw_cmd.c:22:
/ArchBootstraping271217/rtl8723de-4.11-up/include/osdep_service_linux.h: In function ‘_init_timer’:
/ArchBootstraping271217/rtl8723de-4.11-up/include/osdep_service_linux.h:287:8: error: ‘_timer {aka struct timer_list}’ has no member named ‘data’
  ptimer->data = (unsigned long)cntx;
        ^~
/ArchBootstraping271217/rtl8723de-4.11-up/include/osdep_service_linux.h:288:2: error: implicit declaration of function ‘init_timer’; did you mean ‘_init_timer’? [-Werror=implicit-function-declaration]
  init_timer(ptimer);
  ^~~~~~~~~~
  _init_timer
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:317: /ArchBootstraping271217/rtl8723de-4.11-up/core/rtw_cmd.o] Error 1
make[1]: *** [Makefile:1508: _module_/ArchBootstraping271217/rtl8723de-4.11-up] Error 2
make[1]: Leaving directory '/usr/lib/modules/4.15.2-2-ARCH/build'
make: *** [Makefile:1884: modules] Error 2

Fedoro, 2018-02-17:
Fedora 27 kernel 4.15.3-300 64 bits same issue:

DKMS make.log for rtl8723de-5.1.1.8_21285.20171026_COEX20170111-1414 for kernel 4.15.3-300.fc27.x86_64 (x86_64)
sáb feb 17 15:53:11 -03 2018
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/4.15.3-300.fc27.x86_64/build M=/var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build  modules 
make[1]: se entra en el directorio '/usr/src/kernels/4.15.3-300.fc27.x86_64'
  CC [M]  /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/core/rtw_cmd.o
In file included from /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/include/osdep_service.h:42:0,
                 from /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/include/drv_types.h:32,
                 from /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/core/rtw_cmd.c:22:
/var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/include/osdep_service_linux.h: En la función ‘_init_timer’:
/var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/include/osdep_service_linux.h:287:8: error: ‘_timer {también conocido como struct timer_list}’ no tiene un miembro llamado ‘data’
  ptimer->data = (unsigned long)cntx;
        ^~  
/var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/include/osdep_service_linux.h:288:2: error: implicit declaration of function ‘init_timer’; did you mean ‘_init_timer’? [-Werror=implicit-function-declaration]
  init_timer(ptimer);
  ^~~~~~~~~~
  _init_timer
cc1: algunos avisos se tratan como errores
make[2]: *** [scripts/Makefile.build:317: /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/core/rtw_cmd.o] Error 1
make[1]: *** [Makefile:1512: _module_/var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build] Error 2
make[1]: se sale del directorio '/usr/src/kernels/4.15.3-300.fc27.x86_64'
make: *** [Makefile:1884: modules] Error 2

ayanb280, 2018-02-23:
I am also facing the same issue on kernel 4.15.3-300.fc27.x86_64 (Fedora 27).

procedural, 2018-02-25:
Linux 4.15 deprecated init_timer

drunkfoxx, 2018-03-05:
So I guess that module is not working (yet?) for all 4.15+ kernels, also tested on gentoo with 4.15.7-r1 kernel, make failed:
[](https://paste.pound-python.org/show/viLExJqlW7nPMdekEQkJ/)

alukin, 2018-03-09:
The problem caused by changed timer interface in 4.15

drunkfoxx, 2018-03-09:
Got it already; is it possible to patch the driver to work with this interface in newer kernels? (I personally am not competent in C)

kenbenobi666, 2018-03-12:
Same problem. 4.15.7-300.fc27.x86_64

anthraxn8b, 2018-03-17:
Same problem on 4.15.9-300.fc27.x86_64.
@smlinux Can you please take a look at that "nice" interface change?
It would make me very happy :-)
THX in advance ;-)

agent00S, 2018-03-17:
Same problem with kernel 4.15.7. When for a new version

mujeebcpy, 2018-03-31:
same problem for ubuntu 18.04 development version which uses 4.15 as kernal

alphacentauri82, 2018-04-27:
@anthraxn8b @agent00S @mujeebcpy Use https://github.com/lwfinger/rtlwifi_new/tree/extended and when you cd in the folder make sure you checkout in the extended branch. It works like a charm

c4explosive, 2018-05-06:
@alphacentauri82 It's compiles and installs, but is not working good. Problems with driver rtl8723de and Kernel Linux 4.16. I had to downgrade to the linux 4.14.15  again.

alphacentauri82, 2018-05-07:
My bad! but downgrading to 4.14.15 sorted out the problem?

c4explosive, 2018-05-08:
Yes, downgrading to 4.14.15 solved the problem.

smlinux, 2018-06-12:
The branch 4.15-up is added to the repository. Try it.

mujeebcpy, 2018-06-12:
thanks.. it works for 4.15,
got no found kernal error in 4.16

c4explosive, 2018-11-08:
Yes, the branch 4.15-up is working. Thanks.