smlinux/rtl8723de #14: Support Linux Kernel 4.9

rishabhtss, 2018-03-09:
Any chance of supporting Linux Kernel 4.9?
Debian Strech(Stable) latest Kernel is only 4.9.
These Drivers require 4.11 I don't want to upgrade and mess up other packages from stable Debian stable repository.

ayanb280, 2018-03-09:
Have you tried installing branch [4.10-down](https://github.com/smlinux/rtl8723de/tree/4.10-down)?

rishabhtss, 2018-03-09:
Yes now i tried down version but,still not luck.
on running **make**
```
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/4.9.0-4-amd64/build M=/home/debian/Documents/rtl8723de  modules
make[1]: Entering directory '/usr/src/linux-headers-4.9.0-4-amd64'
  CC [M]  /home/debian/Documents/rtl8723de/core/rtw_cmd.o
  CC [M]  /home/debian/Documents/rtl8723de/core/rtw_security.o
  CC [M]  /home/debian/Documents/rtl8723de/core/rtw_debug.o
  CC [M]  /home/debian/Documents/rtl8723de/core/rtw_io.o
  CC [M]  /home/debian/Documents/rtl8723de/core/rtw_ioctl_query.o
  CC [M]  /home/debian/Documents/rtl8723de/core/rtw_ioctl_set.o
  CC [M]  /home/debian/Documents/rtl8723de/core/rtw_ieee80211.o
  CC [M]  /home/debian/Documents/rtl8723de/core/rtw_mlme.o
  CC [M]  /home/debian/Documents/rtl8723de/core/rtw_mlme_ext.o
  CC [M]  /home/debian/Documents/rtl8723de/core/rtw_mi.o
  CC [M]  /home/debian/Documents/rtl8723de/core/rtw_wlan_util.o
  CC [M]  /home/debian/Documents/rtl8723de/core/rtw_vht.o
  CC [M]  /home/debian/Documents/rtl8723de/core/rtw_pwrctrl.o
  CC [M]  /home/debian/Documents/rtl8723de/core/rtw_rf.o
  CC [M]  /home/debian/Documents/rtl8723de/core/rtw_recv.o
  CC [M]  /home/debian/Documents/rtl8723de/core/rtw_sta_mgt.o
  CC [M]  /home/debian/Documents/rtl8723de/core/rtw_ap.o
  CC [M]  /home/debian/Documents/rtl8723de/core/rtw_xmit.o
  CC [M]  /home/debian/Documents/rtl8723de/core/rtw_p2p.o
  CC [M]  /home/debian/Documents/rtl8723de/core/rtw_tdls.o
  CC [M]  /home/debian/Documents/rtl8723de/core/rtw_br_ext.o
  CC [M]  /home/debian/Documents/rtl8723de/core/rtw_iol.o
  CC [M]  /home/debian/Documents/rtl8723de/core/rtw_sreset.o
  CC [M]  /home/debian/Documents/rtl8723de/core/rtw_btcoex.o
  CC [M]  /home/debian/Documents/rtl8723de/core/rtw_beamforming.o
  CC [M]  /home/debian/Documents/rtl8723de/core/rtw_odm.o
  CC [M]  /home/debian/Documents/rtl8723de/core/efuse/rtw_efuse.o
  CC [M]  /home/debian/Documents/rtl8723de/os_dep/osdep_service.o
  CC [M]  /home/debian/Documents/rtl8723de/os_dep/linux/os_intfs.o
  CC [M]  /home/debian/Documents/rtl8723de/os_dep/linux/pci_intf.o
  CC [M]  /home/debian/Documents/rtl8723de/os_dep/linux/pci_ops_linux.o
  CC [M]  /home/debian/Documents/rtl8723de/os_dep/linux/ioctl_linux.o
  CC [M]  /home/debian/Documents/rtl8723de/os_dep/linux/xmit_linux.o
  CC [M]  /home/debian/Documents/rtl8723de/os_dep/linux/mlme_linux.o
  CC [M]  /home/debian/Documents/rtl8723de/os_dep/linux/recv_linux.o
  CC [M]  /home/debian/Documents/rtl8723de/os_dep/linux/ioctl_cfg80211.o
  CC [M]  /home/debian/Documents/rtl8723de/os_dep/linux/rtw_cfgvendor.o
  CC [M]  /home/debian/Documents/rtl8723de/os_dep/linux/wifi_regd.o
  CC [M]  /home/debian/Documents/rtl8723de/os_dep/linux/rtw_android.o
  CC [M]  /home/debian/Documents/rtl8723de/os_dep/linux/rtw_proc.o
  CC [M]  /home/debian/Documents/rtl8723de/os_dep/linux/ioctl_mp.o
  CC [M]  /home/debian/Documents/rtl8723de/hal/hal_intf.o
  CC [M]  /home/debian/Documents/rtl8723de/hal/hal_com.o
  CC [M]  /home/debian/Documents/rtl8723de/hal/hal_com_phycfg.o
  CC [M]  /home/debian/Documents/rtl8723de/hal/hal_phy.o
  CC [M]  /home/debian/Documents/rtl8723de/hal/hal_dm.o
  CC [M]  /home/debian/Documents/rtl8723de/hal/hal_btcoex.o
  CC [M]  /home/debian/Documents/rtl8723de/hal/hal_mp.o
  CC [M]  /home/debian/Documents/rtl8723de/hal/hal_mcc.o
  CC [M]  /home/debian/Documents/rtl8723de/hal/hal_hci/hal_pci.o
  CC [M]  /home/debian/Documents/rtl8723de/hal/led/hal_pci_led.o
  CC [M]  /home/debian/Documents/rtl8723de/hal/HalPwrSeqCmd.o
  CC [M]  /home/debian/Documents/rtl8723de/hal/rtl8723d/Hal8723DPwrSeq.o
  CC [M]  /home/debian/Documents/rtl8723de/hal/rtl8723d/rtl8723d_sreset.o
  CC [M]  /home/debian/Documents/rtl8723de/hal/rtl8723d/rtl8723d_hal_init.o
/home/debian/Documents/rtl8723de/hal/rtl8723d/rtl8723d_hal_init.c: In function ‘SetHwReg8723D’:
/home/debian/Documents/rtl8723de/hal/rtl8723d/rtl8723d_hal_init.c:5359:6: warning: this ‘else’ clause does not guard... [-Wmisleading-indentation]
    } else
      ^~~~
/home/debian/Documents/rtl8723de/hal/rtl8723d/rtl8723d_hal_init.c:5362:5: note: ...this statement, but the latter is misleadingly indented as if it is guarded by the ‘else’
     ulCommand = CAM_CONTENT_COUNT * ucIndex + i;
     ^~~~~~~~~
  CC [M]  /home/debian/Documents/rtl8723de/hal/rtl8723d/rtl8723d_phycfg.o
  CC [M]  /home/debian/Documents/rtl8723de/hal/rtl8723d/rtl8723d_rf6052.o
  CC [M]  /home/debian/Documents/rtl8723de/hal/rtl8723d/rtl8723d_dm.o
[... 61 lines cut ...]
  CC [M]  /home/debian/Documents/rtl8723de/hal/phydm/rtl8723d/phydm_regconfig8723d.o
  CC [M]  /home/debian/Documents/rtl8723de/hal/phydm/rtl8723d/phydm_rtl8723d.o
  CC [M]  /home/debian/Documents/rtl8723de/hal/phydm/rtl8723d/halphyrf_8723d.o
/home/debian/Documents/rtl8723de/hal/phydm/rtl8723d/halphyrf_8723d.c: In function ‘PHY_IQCalibrate_8723D’:
/home/debian/Documents/rtl8723de/hal/phydm/rtl8723d/halphyrf_8723d.c:3301:5: warning: this ‘for’ clause does not guard... [-Wmisleading-indentation]
     for (i = 0; i < 8; i++)
     ^~~
/home/debian/Documents/rtl8723de/hal/phydm/rtl8723d/halphyrf_8723d.c:3304:6: note: ...this statement, but the latter is misleadingly indented as if it is guarded by the ‘for’
      if (RegTmp != 0)
      ^~
/home/debian/Documents/rtl8723de/hal/phydm/rtl8723d/halphyrf_8723d.c:3373:3: warning: this ‘for’ clause does not guard... [-Wmisleading-indentation]
   for (i = 0; i < IQK_Matrix_REG_NUM; i++)
   ^~~
[... cut ...]
    pDM_Odm->RFCalibrateInfo.IQKMatrixRegSetting[Indexforchannel].bIQKDone = TRUE;  
    ^~~~~~~
  CC [M]  /home/debian/Documents/rtl8723de/platform/platform_ops.o
  CC [M]  /home/debian/Documents/rtl8723de/core/rtw_mp.o
  LD [M]  /home/debian/Documents/rtl8723de/8723de.o
  Building modules, stage 2.
  MODPOST 1 modules
  CC      /home/debian/Documents/rtl8723de/8723de.mod.o
  LD [M]  /home/debian/Documents/rtl8723de/8723de.ko
make[1]: Leaving directory '/usr/src/linux-headers-4.9.0-4-amd64'
```