smlinux/rtl8723de #15: Support for Kali Linux

alvaara, 2018-03-26:
Hi! I was installing rtl8723 for my recent single OS installation of kali wirh kernel >= 14.11 as stated by output for:
uname -r

> 4.14.0-kali3-amd64

cat /proc/version

> Linux version 4.14.0-kali3-amd64 (devel@kali.org) (gcc version 7.2.0 (Debian 7.2.0-18)) #1 SMP Debian 4.14.12-2kali1 (2018-01-08)

but upon reaching step:

_dkms install rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414_

I got the following error:

> Kernel preparation unnecessary for this kernel.  Skipping...
> 
> Building module:
> cleaning build area...
> 'make' all KVER=4.14.0-kali3-amd64......(bad exit status: 2)
> Error! Bad return status for module build on kernel: 4.14.0-kali3-amd64 (x86_64)
>  Consult /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/make.log for more information.

make.log reads:

> DKMS make.log for rtl8723de-5.1.1.8_21285.20171026_COEX20170111-1414 for kernel 4.14.0-kali3-amd64 (x86_64)
> Mon Mar 26 02:45:41 -03 2018
> /bin/sh: 1: bc: not found
> make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/4.14.0-kali3-amd64/build M=/var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build  modules
> make[1]: Entering directory '/usr/src/linux-headers-4.14.0-kali3-amd64'
> /bin/sh: 1: bc: not found
>   CC [M]  /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/core/rtw_cmd.o
>   CC [M]  /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/core/rtw_security.o
>   CC [M]  /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/core/rtw_debug.o
> /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/core/rtw_debug.c: In function ‘dump_drv_version’:
> /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/core/rtw_debug.c:50:62: error: macro "__DATE__" might prevent reproducible builds [-Werror=date-time]
>   RTW_PRINT_SEL(sel, "build time: %s %s\n", __DATE__, __TIME__);
>                                                               ^
> /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/core/rtw_debug.c:50:62: error: macro "__TIME__" might prevent reproducible builds [-Werror=date-time]
> /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/core/rtw_debug.c:50:62: error: macro "__DATE__" might prevent reproducible builds [-Werror=date-time]
> /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/core/rtw_debug.c:50:62: error: macro "__TIME__" might prevent reproducible builds [-Werror=date-time]
> cc1: some warnings being treated as errors
> /usr/src/linux-headers-4.14.0-kali3-common/scripts/Makefile.build:325: recipe for target '/var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/core/rtw_debug.o' failed
> make[4]: *** [/var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/core/rtw_debug.o] Error 1
> /usr/src/linux-headers-4.14.0-kali3-common/Makefile:1525: recipe for target '_module_/var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build' failed
> make[3]: *** [_module_/var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build] Error 2
> Makefile:146: recipe for target 'sub-make' failed
> make[2]: *** [sub-make] Error 2
> Makefile:8: recipe for target 'all' failed
> make[1]: *** [all] Error 2
> make[1]: Leaving directory '/usr/src/linux-headers-4.14.0-kali3-amd64'
> Makefile:1884: recipe for target 'modules' failed
> make: *** [modules] Error 2

Would you please help me? I bought this laptop for work but I can't do much without wireless, I would really appreciate your help. Thanks

Ovvlly, 2018-03-29:
Possibly you need to delet old wifi driver before installation new. I have the same problem and i cant fix it(

alvaara, 2018-03-30:
Hi! Thx for your answer. I was able to get mi wifi adapter working, but it gets deactivetade sometimes upon reboot. Do you know why might this be happening? Also since having my wifi enabled, the system stalls on boot with a start job for about five minutes. I could paste an image for that when it happens again. I'm thinking on a fresh reinstall and make sure I follow dkms installation steps.

Regarding your answer, how can I delete the old driver? I don't really know where default files are stored I'm not that advanced of a linux user yet. Thank you!

Ovvlly, 2018-04-02:
So the problem is not in the old drivers. I fix this simply. I just downgrade my kernel from 4.14 to 4.9 and it worked! U can do it too  or update ur kernel or u can configure and compile ur own kernel with 4.4 ..... source code. So u need do smth with ur kernel and and everything will be okay!

alvaara, 2018-04-04:
Yes! I actually solved it by installing linux-image for 4.15 and also installing headers for that kernel version, now everything is fixed! Though it seems to crash on startup but I have to disable and enable it a few times for it to work. Thanks for the helping hand!

Ovvlly, 2018-04-05:
Np dude!