smlinux/rtl8723de #17: Doesn't work after successful installation

lordkeks, 2018-04-09:
So I cloned and installed as the readme told. No errors during installation, but there is still no wifi interface after reboot.

Machine is an hp 14-bp003ng with the mentioned rtl8723de chip, running a fresh install of ubuntu 17.10.
If you need any more information, let me know.

If it helps, here is the `lspci` output:
```text
00:00.0 Host bridge: Intel Corporation Xeon E3-1200 v6/7th Gen Core Processor Host Bridge/DRAM Registers (rev 02)
00:02.0 VGA compatible controller: Intel Corporation HD Graphics 620 (rev 02)
00:04.0 Signal processing controller: Intel Corporation Skylake Processor Thermal Subsystem (rev 02)
00:14.0 USB controller: Intel Corporation Sunrise Point-LP USB 3.0 xHCI Controller (rev 21)
00:14.2 Signal processing controller: Intel Corporation Sunrise Point-LP Thermal subsystem (rev 21)
00:16.0 Communication controller: Intel Corporation Sunrise Point-LP CSME HECI #1 (rev 21)
00:17.0 SATA controller: Intel Corporation Sunrise Point-LP SATA Controller [AHCI mode] (rev 21)
00:1c.0 PCI bridge: Intel Corporation Sunrise Point-LP PCI Express Root Port (rev f1)
00:1c.4 PCI bridge: Intel Corporation Sunrise Point-LP PCI Express Root Port #5 (rev f1)
00:1c.5 PCI bridge: Intel Corporation Sunrise Point-LP PCI Express Root Port #6 (rev f1)
00:1f.0 ISA bridge: Intel Corporation Device 9d4e (rev 21)
00:1f.2 Memory controller: Intel Corporation Sunrise Point-LP PMC (rev 21)
00:1f.3 Audio device: Intel Corporation Device 9d71 (rev 21)
00:1f.4 SMBus: Intel Corporation Sunrise Point-LP SMBus (rev 21)
02:00.0 Network controller: Realtek Semiconductor Co., Ltd. Device d723
03:00.0 Ethernet controller: Realtek Semiconductor Co., Ltd. RTL8111/8168/8411 PCI Express Gigabit Ethernet Controller (rev 15)
```

kylemartin87, 2018-05-22:
I was going crazy trying to get the wifi working on Ubuntu 18.04 with the same wifi module (rtl8723de). Somewhere I stumbled across a post that mentioned to **disable secure boot in the bios**. Problem solved!

andres2490, 2018-06-23:
@kylemartin87  How you can fix this problem on Ubuntu 18.04?
I have xubuntu with the kernel 4.15.0-23-generic and it dont works 
I have a Hp computer too (14-bp001)

GringoPbSr, 2019-09-24:
I found temporary solution. Before Ubuntu starts go to **Advanced option for Ubuntu** and choose **Ubuntu, with Linux 4.18.0.25-generic**