smlinux/rtl8723de #19: Issue with 18.04

yogeshrb, 2018-04-26:
I just updated my laptop to 18.04 and it doesn't detects the Wi-Fi card 
the 4.11 and up version doesn't work with 18.04

alphacentauri82, 2018-04-27:
@yogeshrb did you fix it? i can't make it work either.
```bash
Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
'make' all KVER=4.15.0-20-generic....(bad exit status: 2)
ERROR (dkms apport): binary package for rtl8723de: 5.1.1.8_21285.20171026_COEX20170111-1414 not found
Error! Bad return status for module build on kernel: 4.15.0-20-generic (x86_64)
Consult /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/make.log for more information.
```
And the log
```bash
DKMS make.log for rtl8723de-5.1.1.8_21285.20171026_COEX20170111-1414 for kernel 4.15.0-20-generic (x86_64)
vie abr 27 14:03:02 -03 2018
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/4.15.0-20-generic/build M=/var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build  modules
make[1]: Entering directory '/usr/src/linux-headers-4.15.0-20-generic'
Makefile:976: "Cannot use CONFIG_STACK_VALIDATION=y, please install libelf-dev, libelf-devel or elfutils-libelf-devel"
  CC [M]  /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/core/rtw_cmd.o
In file included from /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/include/osdep_service.h:42:0,
                 from /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/include/drv_types.h:32,
                 from /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/core/rtw_cmd.c:22:
/var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/include/osdep_service_linux.h: In function ‘_init_timer’:
/var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/include/osdep_service_linux.h:287:8: error: ‘_timer {aka struct timer_list}’ has no member$
  ptimer->data = (unsigned long)cntx;
        ^~
/var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/include/osdep_service_linux.h:288:2: error: implicit declaration of function ‘init_timer’;$
  init_timer(ptimer);
  ^~~~~~~~~~
  _init_timer
cc1: some warnings being treated as errors
scripts/Makefile.build:332: recipe for target '/var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/core/rtw_cmd.o' failed
make[2]: *** [/var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/core/rtw_cmd.o] Error 1
Makefile:1552: recipe for target '_module_/var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build' failed
make[1]: *** [_module_/var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-4.15.0-20-generic'
Makefile:1884: recipe for target 'modules' failed
make: *** [modules] Error 2
```

lordkeks, 2018-04-27:
@alphacentauri82 
The output means that make could not compile the module. The log shows that it's due to a changed timer interface. That's a problem with ubuntu. Try a different linux distribution, with kernel 4.11 or higher.

alphacentauri82, 2018-04-27:
@lordkeks It's ubuntu 18.04 LTS  Kernel 4.15.0-20

anshulgoyal15, 2018-04-27:
Try this :https://github.com/lwfinger/rtlwifi_new/tree/extended?files=1
Worked with kernel 4.16.2