smlinux/rtl8723de #23: kernal header not found

faisal29, 2018-07-16:
Hello, I get an error on the code : dkms install rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414

Error! echo
Your kernel headers for kernel 4.17.6-1.el7.elrepo.x86_64 cannot be found at
/lib/modules/4.17.6-1.el7.elrepo.x86_64/build or /lib/modules/4.17.6-1.el7.elrepo.x86_64/source.

Help.......

gavindsouza, 2019-01-04:
kernel headers have to be manually installed, but can be easily done with the help of your package manager.

Dhirajraje, 2020-06-08:
If you are on the arch based system, try installing appropriate linux-headers with
`sudo pacman -S linux-headers`
know your kernel by
`hostnamectl`
eg. if your kernel is linux54 i.e. Linux 5.4 install linux54-headers by 
`sudo pacman -S linux54-headers`
in my case
`
┌─[dhirajraje][Darth-Vader][~]
└─▪  hostnamectl
   Static hostname: Darth-Vader
         Icon name: computer-laptop
           Chassis: laptop
        Machine ID: a2e0d36ae62444c891c9b316d3c3f15f
           Boot ID: f900fadba8fd494cac84f334f7749e3b
  Operating System: Manjaro Linux
            Kernel: Linux 5.6.15-1-MANJARO
      Architecture: x86-64

┌─[dhirajraje][Darth-Vader][~]
└─▪ 
`
its **linux56-headers**

closed 2020-06-21 by faisal-ops