smlinux/rtl8723de #27: Fails to build using 4.19 kernel

Fuzzbox999, 2018-10-23:
Hi,

Ubuntu 18.04.1, mainline kernels, dkms. 

Thank you for this driver which saved my HP laptop...

Everything was working flawlessly, and I had no problem upgrading several times from kernel 4.15 to latest versions including 4.18.16.

However building the driver fails against 4.19 with the following log : 

In file included from /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/include/osdep_service_linux.h:86:0,
                 from /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/include/osdep_service.h:42,
                 from /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/include/drv_types.h:32,
                 from /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/os_dep/linux/os_intfs.c:22:
./include/linux/ieee80211.h:1442:0: note: this is the location of the previous definition
 #define IEEE80211_MAX_AMPDU_BUF  0x100

/var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/os_dep/linux/os_intfs.c:1210:22: error: initialization from incompatible pointer type [-Werror=incompatible-pointer-types]
  .ndo_select_queue = rtw_select_queue,
                      ^~~~~~~~~~~~~~~~
/var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/os_dep/linux/os_intfs.c:1210:22: note: (near initialization for ‘rtw_netdev_ops.ndo_select_queue’)
cc1: some warnings being treated as errors
scripts/Makefile.build:305: recipe for target '/var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/os_dep/linux/os_intfs.o' failed
make[2]: *** [/var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/os_dep/linux/os_intfs.o] Error 1
Makefile:1517: recipe for target '_module_/var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build' failed
make[1]: *** [_module_/var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build] Error 2
make[1] : on quitte le répertoire « /usr/src/linux-headers-4.19.0-041900-generic »
Makefile:1884: recipe for target 'modules' failed
make: *** [modules] Error 2

lghamie, 2018-11-06:
I'm facing the same issue on  4.18.17

Boogie3D, 2018-11-10:
Just a temporary workaround, but you can still compile and install the driver if you add the following line to the top of the Makefile:

`USER_EXTRA_CFLAGS=-Wno-error=incompatible-pointer-types`

It seems to be working normally on 4.19.1-1 as far as I can tell.

sustmi, 2018-11-20:
The interface of `ndo_select_queue` callback changed.
See this commit: https://github.com/torvalds/linux/commit/4f49dec9075aa0277b8c9c657ec31e6361f88724 .
It looks like you only need to change the `rtw_select_queue` signature because the removed `accel_priv` does not look to be used it the code anyway.

You can use a patch similar to this one:
```patch
diff --git a/os_dep/linux/os_intfs.c b/os_dep/linux/os_intfs.c
index 660bacf..2d073bc 100644
--- a/os_dep/linux/os_intfs.c
+++ b/os_dep/linux/os_intfs.c
@@ -757,7 +757,11 @@ unsigned int rtw_classify8021d(struct sk_buff *skb)
  
 static u16 rtw_select_queue(struct net_device *dev, struct sk_buff *skb
 #if LINUX_VERSION_CODE >= KERNEL_VERSION(3, 13, 0)     
+       #if (LINUX_VERSION_CODE >= KERNEL_VERSION(3, 13, 0) && LINUX_VERSION_CODE < KERNEL_VERSION(4, 19, 0))
                                , void *accel_priv
+       #else
+                               , struct net_device *sb_dev     
+       #endif
 #if LINUX_VERSION_CODE >= KERNEL_VERSION(3, 14, 0) 
                                , select_queue_fallback_t fallback
 #endif
```

Even though disabling the type checks would work too in this case, I discourage you to do it as it would make it harder to debug the code in the future.

AstroDrabb, 2019-03-25:
I've had this issue with the drivers for my D-Link DWA-171.  I found one that worked but then stopped around 4.19 or so.  I have it working on 4.19.* and the 5.0.* kernels.
https://github.com/AstroDrabb/rtl8812au

Step to install:
--------------------------------------------------------------
git clone https://github.com/AstroDrabb/rtl8812au
sudo cp -r rtl8812au /usr/src/
sudo dkms add -m rtl8812au -v 4.2.2
sudo dkms build -m rtl8812au -v 4.2.2 -k $(uname -r) 
sudo dkms install -m rtl8812au -v 4.2.2 -k $(uname -r)
--------------------------------------------------------------
That should be it.

Lynaex, 2019-07-01:
Me ha funcionado en OpenSuse Leap 15.1, añadiendo al Makefile  de **AsstroDrabb** la línea de **Boogie 3D**:

 USER_EXTRA_CFLAGS=-Wno-error=incompatible-pointer-types

En fin, tal como el indica parece una solución un tanto a lo espadadrapo, pero tras muchas historias la que me ha funcionado. Indica el error pero sigue adelante y termina.

Muchas gracias a ambos.