smlinux/rtl8723de #30: driver stops working when wake up after suspend

ensarman, 2018-12-16:
as it, the dirver stops working after waing up from suspend or hibernate

modprobe rtl8723de ; lsmod |grep rtl
btrtl                  16384  1 btusb
bluetooth             626688  39 btrtl,btintel,btbcm,bnep,btusb,rfcomm

it seems it only load the driver but doesn't appear in lsmod, it's strange

vladtcvs, 2019-01-01:
I have 8723de, not rtl8723de, in lsmod

vladtcvs, 2019-01-01:
I alse have problem with hibernate.

reproducing:
1) modprobe 8723de
2) connect to wifi
3) suspend
4) wake up
5) try to connect to any wifi (I'm using network manager)

8723de kernel module present in modules

expected result:
wifi network connected

result:
no connection.

I've tested with another wifi modules, they works fine.

With 8723de driver I have to

```
rmmod 8723de
modprobe 8723de
```
After this I can connect to wifi

Mathisca, 2019-05-08:
I have the same issue

shajman, 2019-07-10:
Has anyone found a solution for this? :(

ensarman, 2019-07-10:
My solution was unload and load the driver after wakeup

Joado97, 2019-08-08:
> […](#)

Hi, I'm having the same issue. How do you do that? I'm new with Linux hahahaha

MariaSpethmann, 2019-08-16:
I had the same problem and also the solution was to unload and load the driver, although the name of the driver for unloading was 8723de and for loading it was rtl8723de. Also, unloading via modprobe -r did not work.

https://askubuntu.com/questions/1165894/wireless-network-connection-fails-after-suspending/1166295#1166295

ensarman, 2019-08-17:
I use rmmod instead modprobe -r, but both commands should have the same
effect

alanfox2000, 2019-09-18:
Official Driver Update:
[20190917_RTL8723DE_WiFi_linux_v5.7.7_33046_COEX20181130-2e2e.20190917.zip](https://github.com/alanfox2000/realtek-linux/blob/master/RTL8723DE/20190917_RTL8723DE_WiFi_linux_v5.7.7_33046_COEX20181130-2e2e.20190917.zip)

indieSAG, 2019-09-28:
```
Official Driver Update:
20190917_RTL8723DE_WiFi_linux_v5.7.7_33046_COEX20181130-2e2e.20190917.zip
```
How to install it? It shows error while I run install.sh

serhumanos, 2020-04-29:
I have this problem too.
plus, when i logout and login the wifi present the same behaivor.

(i want to add, that this driver is the only that it can use wifi and bluetooth simultaneous)

Pretuer, 2020-09-06:
> [20190917_RTL8723DE_WiFi_linux_v5.7.7_33046_COEX20181130-2e2e.20190917.zip](https://github.com/alanfox2000/realtek-linux/blob/master/RTL8723DE/20190917_RTL8723DE_WiFi_linux_v5.7.7_33046_COEX20181130-2e2e.20190917.zip)

out dated and 404 on page