smlinux/rtl8723de #32: Failed to build using 4.20 kernel

gavindsouza, 2019-01-01:
hey guys,
kernel: 4.20.0-arch1-1-ARCH 

Everything worked just fine until now. Faced no issues through the past builds. 

the last lines of the make log file 

> /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/make.log

```
/var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/os_dep/linux/ioctl_cfg80211.c: In function ‘rtw_get_systime_us’:
 /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/os_dep/linux/ioctl_cfg80211.c:352:2: error: implicit declaration of function ‘get_monotonic_boottime’; did you mean ‘getboottime’? [-Werror=implicit-function-declaration]
   get_monotonic_boottime(&ts);
   ^~~~~~~~~~~~~~~~~~~~~~
   getboottime
 cc1: some warnings being treated as errors
 make[2]: *** [scripts/Makefile.build:292: /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/os_dep/linux/ioctl_cfg80211.o] Error 1
 make[1]: *** [Makefile:1563: _module_/var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build] Error 2
 make[1]: Leaving directory '/usr/lib/modules/4.20.0-arch1-1-ARCH/build'
 make: *** [Makefile:1885: modules] Error 2
```

AdamKandur, 2019-01-01:
Hi guys, the same problem, please help

gavindsouza, 2019-01-02:
I've opened a PR that fixes the issue for the same

cesarpose, 2019-01-03:
The same problem with linux-ck kernel 4.20

AdamKandur, 2019-01-03:
@cesarpose gavindsouza made a solution. Look his request.

PR #33 (2019-01-04): fix build for Linux-4.20 
fixes #32
closed 2019-01-04 by smlinux