smlinux/rtl8723de #37: Wifi stops working after a period of time

taskmasterREBORN, 2019-02-26:
Wifi stays connected but I can't connect to the internet. Turning it off and then back on temporarily solves this issue. Until this issue is solved I can't download large files on my laptop.

AlejandroFNadal, 2020-07-05:
For all those having this issue, I had to disable automatic suspension. The driver stopped working when the machine went into suspension. If you want to keep going into suspension, do this after you wake it up
`rmmod 8723de
modprobe rtl8723de`

That sort of works. But I have some issues with this driver too, the internet connection fails a bit randomly. I am thinking of going back to the rtl driver or check if I can use the one in the 5.x kernel.

brthngnsh, 2020-07-17:
Ubuntu 20.04LTS now have rtl drivers by default . Try it out . Works fine.