smlinux/rtl8723de #38: Build failed using Linux 5.0 kernel

gavindsouza, 2019-03-07:
## Build environment: 5.0.0-arch1-1-ARCH

the last lines of the make log file 

> /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/make.log

``` 
In file included from /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/include/osdep_service_linux.h:86,
                 from /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/include/osdep_service.h:42,
                 from /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/include/drv_types.h:32,
                 from /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/os_dep/linux/rtw_android.c:25:
./include/linux/ieee80211.h:1444: note: this is the location of the previous definition
 #define IEEE80211_MAX_AMPDU_BUF  0x100
 
/var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/os_dep/linux/rtw_android.c: In function ‘rtw_android_priv_cmd’:
/var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/os_dep/linux/rtw_android.c:629:62: error: macro "access_ok" passed 3 arguments, but takes just 2
  if (!access_ok(VERIFY_READ, priv_cmd.buf, priv_cmd.total_len)) {
                                                              ^
/var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/os_dep/linux/rtw_android.c:629:7: error: ‘access_ok’ undeclared (first use in this function)
  if (!access_ok(VERIFY_READ, priv_cmd.buf, priv_cmd.total_len)) {
       ^~~~~~~~~
/var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/os_dep/linux/rtw_android.c:629:7: note: each undeclared identifier is reported only once for each function it appears in
make[2]: *** [scripts/Makefile.build:277: /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/os_dep/linux/rtw_android.o] Error 1
make[1]: *** [Makefile:1553: _module_/var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build] Error 2
make[1]: Leaving directory '/usr/lib/modules/5.0.0-arch1-1-ARCH/build'
make: *** [Makefile:1885: modules] Error 2
```

n37w1z, 2019-03-08:
yeeee, you also tried to comment some lines in code =)
Well, until they will not do upgrade, you may downgrade your kernel with https://archive.archlinux.org/packages/
you need to download linux-4.15.8-1-x86_64.pkg.tar.xz and linux-headers-4.15.8-1-x86_64.pkg.tar.xz (of course, you may choose another version).
and use it 
sudo pacman -U linux-4.15.8-1-x86_64.pkg.tar.xz linux-headers-4.15.8-1-x86_64.pkg.tar.xz

of course, it's unsafe =)

n37w1z, 2019-03-08:
and dont work =(

n37w1z, 2019-03-08:
ohh, works!

n37w1z, 2019-03-08:
nope

cloudyql, 2019-03-12:
Thank you so much. Applied the patch to OpenSUSE 15.0 running the latest "Kernel_Stable" 5.0.1 and the module finally compiles correctly.
As this version of the module seems to behave far better on my system than others, I am especially grateful.

closed 2019-03-13 by smlinux

fi12ds07, 2019-03-16:
this totally worked for me!! Thank you so much 

> and dont work =(

i do not understand why it didn't work for you :(
maybe you are not doing it right

PR #39 (2019-03-13): fix build for Linux-5.0
fixes #38