smlinux/rtl8723de #43: Linux Kernel 5.1.3 Build Error

noel3225, 2019-05-20:
It fails to build on Linux Kernel 5.1.3. The linux-headers are all installed, it returns 

> bad exit status:2

> Building module:
> cleaning build area...
> 'make' all KVER=5.1.3-arch1-1-ARCH............(bad exit status: 2)
> Error! Bad return status for module build on kernel: 5.1.3-arch1-1-ARCH (x86_64)
> Consult /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/make.log for more information.

**Log File**
`https://pastebin.com/d4HdQ05S`

SuperBoii123, 2019-05-22:
Had the same problem with kernel 5.1.2, also using Arch Linux

seebeck88, 2019-06-02:
I had the same problem and this is what works for me : 
if you don't specify the kernel version it automatic gets the latest version     

    git clone https://github.com/smlinux/rtl8723de.git
    sudo dkms add ./rtl8723de
    sudo dkms install rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414

antonkharkov, 2019-11-03:
Solution found on kernel 5.3.8-arch1-1. Just execute the strings below one by one in the terminal window.
git clone -b extended https://github.com/lwfinger/rtlwifi_new.git
Here remove the Wifi dongle unless you have a wired connection and continue.
sudo dkms add ./rtlwifi_new
sudo dkms install rtlwifi-new/0.6
sudo cp -r rtlwifi_new/firmware/rtlwifi/ /lib/firmware/rtlwifi
echo "options rtl8723de ant_sel=2" | sudo tee /etc/modprobe.d/rtl8723de.conf
reboot
After rebooting, all works fine. Just find your network and input the password. Enjoy!