smlinux/rtl8723de #44: Drivers won't work, even if completely installed

mangoppola, 2019-05-25:
I followed the instructions and installed the drivers from the 5.0-up branch on my KDE manjaro system, with kernel 5.1.1-2. Even though I don't get any errors during installation and `dkms status rtl8723de` reports `rtl8723de, 5.1.1.8_21285.20171026_COEX20170111-1414, 5.1.1-2-MANJARO, x86_64: installed`, I cannot activate neither wifi nor bluetooth, and in the "hardware configuration" menu, the network card will be displayed as an unknown device. Also, the device shows as RTL8723DE when running `lspci` so I don't really know what is going on

Maxcension, 2019-07-01:
do you found a fix?

mangoppola, 2019-07-01:
I did. I reinstalled the whole OS and the drivers 😂😂

Maxcension, 2019-07-01:
lmaoo
because i have the same issue but i have a fresh install of Ubuntu 19.04 sooo

mangoppola, 2019-07-01:
Are you sure you downloaded from the right branch?

Maxcension, 2019-07-01:
yeah 5.0-up like the readme.md say

mangoppola, 2019-07-01:
You need to match that with your own kernel. I think you should use the 4.19-up branch, but you should check that for yourself

Maxcension, 2019-07-01:
i use ubuntu 19.04 so the kernel version is 5.x so it should work w/ 5.0-up :/

mangoppola, 2019-07-01:
You're right. Sorry, but I'm no expert. I really can't help you any more than this

Maxcension, 2019-07-01:
a,nyway, thanks a lot!

alanfox2000, 2019-09-18:
[20190917_RTL8723DE_WiFi_linux_v5.7.7_33046_COEX20181130-2e2e.20190917.zip](https://github.com/alanfox2000/realtek-linux/blob/master/RTL8723DE/20190917_RTL8723DE_WiFi_linux_v5.7.7_33046_COEX20181130-2e2e.20190917.zip)

GringoPbSr, 2019-09-24:
I found temporary solution. Before Ubuntu starts go to **Advanced option for Ubuntu** and choose **Ubuntu, with Linux 4.18.0.25-generic**