smlinux/rtl8723de #46: Build fails on xubuntu 18.04

pik314, 2019-07-20:
I have a HP laptop 15t-da000 running windows 10 and xubuntu 18.04. dkms install failed. Here is is the make.log

[make.log](https://github.com/smlinux/rtl8723de/files/3413825/make.log)

gavindsouza, 2019-07-20:
give more information about your environment like linux version and specify even the branch cloned.

pik314, 2019-07-21:
The linux version is in the title (xubuntu 18.04). I thought the branch (master) can be inferred from the log file (rtl8723de-5.1.1.8_21285.20171026_COEX20170111-1414). Let me know if you need anything else.

pik314, 2019-07-25:
I checked and the branch was 5.0-up, if this is of any help.

gavindsouza, 2019-07-25:
> The linux version is in the title (xubuntu 18.04).

that's the distro version. the kernel version can be found using a `uname -r` command

pik314, 2019-07-25:
Isn't the kernel version in the first line of the log file (kernel 4.18.0-25-generic (x86_64) ? What am I missing here?

closed 2019-08-04 by pik314

pik314, 2019-08-04:
Wrong kernel version. Mine was 4.18.0-25-generic. Installed the 4.15-up version of the module and now WiFiworks but the speed is much slower than in Windows