smlinux/rtl8723de #51: kernel panic linux 5.3 Manjaro

tux871, 2019-10-07:
kernel panic on boot. Manjaro 5.3.0-1.

nahuelwexd, 2019-10-16:
Kernel panic here on Arch, since kernel 5.3 and later

zarath, 2019-10-21:
This is caused since linux v5.2 fallback argument is removed from
dev->ndo_select_queue()

Hopefully my 5.2-up merge requests will fix this issue

antonkharkov, 2019-11-02:
Solution found on kernel 5.3.8-arch1-1. Just execute the strings below one by one in the terminal window.
git clone -b extended https://github.com/lwfinger/rtlwifi_new.git
Here remove the Wifi dongle unless you have a wired connection and continue.
sudo dkms add ./rtlwifi_new
sudo dkms install rtlwifi-new/0.6
sudo cp -r rtlwifi_new/firmware/rtlwifi/ /lib/firmware/rtlwifi
echo "options rtl8723de ant_sel=2" | sudo tee /etc/modprobe.d/rtl8723de.conf
reboot
After rebooting, all works fine. Just find your network and input the password. Enjoy!

amstelchen, 2020-02-13:
Confirming RTL8723DE works on 5.4.15-arch1-1 and 5.4.19-1-lts however I had a lot of disconnects with 5.4.15. Did not get it to work with latest 5.5 (non-LTS) kernel resulting in kernel panic.
When updating the kernel,  manual copy into /lib/firmware/rtlwifi was necessary when 'dkms status' noted 'WARNING! Diff between built and installed module!'