smlinux/rtl8723de #52: Build Error Kernel 5.3.X-X

Noryub, 2019-10-07:
Terminal Output:

```
[hp240g6 rtl8723de-5.0-up]# make
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.3.4-1-MANJARO/build M=/home/X/Descargas/rtl8723de-5.0-up  modules
make[1]: se entra en el directorio '/usr/lib/modules/5.3.4-1-MANJARO/build'
  CC [M]  /home/X/Descargas/rtl8723de-5.0-up/core/rtw_cmd.o
cc1: error: incompatible gcc/plugin versions
cc1: error: no se puede inicializar el plugin ./scripts/gcc-plugins/structleak_plugin.so
make[2]: *** [scripts/Makefile.build:281: /home/X/Descargas/rtl8723de-5.0-up/core/rtw_cmd.o] Error 1
make[1]: *** [Makefile:1624: _module_/home/X/Descargas/rtl8723de-5.0-up] Error 2
make[1]: se sale del directorio '/usr/lib/modules/5.3.4-1-MANJARO/build'
make: *** [Makefile:1885: modules] Error 2
``

Noryub, 2019-10-08:
<< UPDATE >>

This error happened to me in the version 5.3.4.1, its a problem in one script of the headers.

But the kernel updated recently to 5.3.5.1, and now this problem has been fixed. The module builds without errors.

(Maybe this happened only with me, I will close this issue, false alarm.)

closed 2019-10-08 by Noryub

gavindsouza, 2019-10-08:
happened with me too, but if it's fixed now 👍🏼