smlinux/rtl8723de #55: Not getting permanently removed

rytvish, 2019-10-19:
I am having issue that bluetooth is not working. So, I want to remove this already installed wifi driver and try installing [rtlwifi_new](https://github.com/lwfinger/rtlwifi_new/tree/extended) but it keeps coming back after reboot, how to permanently remove it. 
You can find some relevant info about my system [here](https://pastebin.com/Qf28RsTb)

DISTRIB_ID=ManjaroLinux
DISTRIB_RELEASE=18.1.1
DISTRIB_CODENAME=Juhraya

[rytvish ~]# lsmod| grep 8723 
8723de               1867776  0
cfg80211              856064  1 8723de
[rytvish ~]# man rmmod
[rytvish ~]# 
[rytvish ~]# 
[rytvish ~]# modprobe -r 8723de
modprobe: FATAL: Module 8723de not found.
[rytvish ~]# 
[rytvish ~]# lsmod| grep 8723 
8723de               1867776  0
cfg80211              856064  1 8723de
[rytvish ~]# rmmod -f 8723de
[rytvish ~]# dkms uninstall rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414
Error! The module rtl8723de 5.1.1.8_21285.20171026_COEX20170111-1414 is not currently installed.
This module is not currently ACTIVE for kernel 5.2.21-1-MANJARO (x86_64).
[rytvish ~]# dkms remove rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414 --all
[rytvish ~]# depmod -a
[rytvish ~]#

Kindly help.

Regards,
Ritesh Vishwakarma

Dhirajraje, 2020-06-08:
After the following process in README, try 
    **_sudo rm -r /var/lib/dkms/rtl8723de*_**

Dhirajraje, 2020-06-08:
After the following process in **README**, try
`sudo rm -r /var/lib/dkms/rtl8723de*
`