smlinux/rtl8723de #57: wifi stops working when trying to switch

maleksiss, 2019-10-26:
My wifi-module works fine and I connect to networks successfully.But if  I want to switch to another Wifi , it doesnt work.
So I need to uninstall and reinstall rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414 and reboot .Then it works!

Kindly help.
Regards,
Malek

ZN0N15, 2019-11-09:
I have  a exact same issue, after booting up my system the first wifi network it scans does connect but after I switch to a different network it would stop working again. Somehow I can only get it to connect to a wifi network after a reboot of the system and once its disconnected, the wifi will not connect unless another reboot is done once again(I dont need to reinstall the drivers). It only works once after each boot of the system, but everything else works, it can detect wifi networks and can connect to any of them but only once(cant switch again to a different network once already connected to one network).

nbarbaz, 2019-11-21:
Same issue with me as well, can't switch to another network without rebooting.

kormang, 2019-11-22:
Thanks for the driver.
Sorry if I'm being rude for asking:
Can we hope that this will be fixed any time soon?

vi3v, 2019-12-03:
And I have... I restart NetworkManager to work.

cata0309, 2020-01-28:
I am experiencing the same issue. Can this be fixed ?

Devansh80, 2020-06-30:
Please take a look on it #57

I have a exact same issue, after booting up my system the first wifi network it scans does connect but after I switch to a different network it would stop working again. Somehow I can only get it to connect to a singke wifi network after a reboot of the system and once its disconnected, the wifi will not connect unless another reboot is done once again(I dont need to reinstall the drivers). It only works once after each boot of the system, but everything else works fine, it can detect wifi networks and can connect to any of them but only once(cant switch again to a different network once its already connected to one network)

DanilChechkov, 2020-06-30:
Hi, thank you driver works fine, but same issue as guys have. Cannot change wifi or after waking from sleep mode cannot connect to wi-fi. Not a big problem, but maybe one day it can be fixed?

Devansh80, 2020-06-30:
Yes hope so

DanilChechkov, 2020-06-30:
> Yes hope so
Also you can try [that one.](https://github.com/rtlwifi-linux/rtw88_8723de) Just installed and all good

ZN0N15, 2020-06-30:
> > Yes hope so
> 
> Also you can try [that one.](https://github.com/rtlwifi-linux/rtw88_8723de) Just installed and all good

Does Bluetooth work on that one? Ive been using [this one](https://github.com/lwfinger/rtlwifi_new/tree/rtw88) but Bluetooth dosnt work

DanilChechkov, 2020-06-30:
Hey Just checked with my Air Pods and I am very surprised - first time during last half of a year I connected them to laptop and even can hear music!. So yep, the answer - bluetooth and wifi work fine

Devansh80, 2020-06-30:
Yes Bluetooth working fine.

ZN0N15, 2020-06-30:
> Hey Just checked with my Air Pods and I am very surprised - first time during last half of a year I connected them to laptop and even can hear music!. So yep, the answer - bluetooth and wifi work fine

Thanks

DanilChechkov, 2020-06-30:
You're welcome guys!=)

ZN0N15, 2020-06-30:
> Hey Just checked with my Air Pods and I am very surprised - first time during last half of a year I connected them to laptop and even can hear music!. So yep, the answer - bluetooth and wifi work fine

What Bluetooth front-end are you using? I can't seem to get bluetooth working on mine. It doesn't seem to detect any bluetooth adapters. Im using bluez and blueman front-ends. Wifi is working fine though.

DanilChechkov, 2020-06-30:
Blueman for gnome, using back end bluez D-bus. What distro u use?

ZN0N15, 2020-06-30:
Nevermind. I just realized that it might possibly be because of my kernel. Im using Arch with mainline kernel (not the LTS one) It's possible that the bluetooth doesn't work with kernel 5.4 and above. Just like this one we are writing on(smlinux).

S4GAR, 2020-09-11:
I am also facing the same issue, Please assist, If anyone knows how to deal with it. 
kernel version: 5.7.0-2parrot2-amd64

Devansh80, 2020-09-11:
I am using Ubuntu 18.04.5 LTS with kernel version: (5.4.0-47) and there is no problem with wifi or bluetooth