smlinux/rtl8723de #66: Please close this repo

vitaliy-kuzmich, 2020-02-28:
Hello, code in this repo broke down the whole system, please remove it. Or mark as out of date.

andrescanm, 2020-02-28:
Yes! Confirmed, I tried several distributions and in all the same problem arises, when restarting the laptop everything goes to hell.

vitaliy-kuzmich, 2020-02-28:
Here it is working one : https://github.com/lwfinger/rtlwifi_new

codejack-CR, 2020-02-29:
> Hello, code in this repo broke down the whole system, please remove it. Or mark as out of date.

On Linux Kernel 5.5, the startup messages show linux kernel panic. This repo is dangerous for anyone trying to use kernel version > 5.3.

mikhailnov, 2020-03-28:
What is the problem exactly?

codejack-CR, 2020-03-29:
> What is the problem exactly?

I don't have the exact log but, I remember the error messages being that the installed version of the driver differs from the kernel version and that it failed to load for that reason. You can test it on a separate partitioned OS if you have the same wifi module.

mikhailnov, 2020-03-29:
> that the installed version of the driver differs from the kernel version and that it failed to load for that reason

My patch https://github.com/smlinux/rtl8723de/pull/67 fixes it

rongx1an, 2020-04-09:
This driver is perfect on xubuntu 19.04 which using the kernel version 5.0.

adrianperal, 2020-04-27:
today i updated manjaro to 20.0 with lts kernel 5.4, went smooth, no problem at all, everything working as it should. RTL8723de

inpos, 2020-04-30:
DON'T close this repo. Only this driver is magically working. Driver from "rtlwifi_new" repo drop me to hell.
Thank you for this repo.

vitaliy-kuzmich, 2020-04-30:
okay, 2 confirmations that it works

closed 2020-04-30 by vitaliy-kuzmich