smlinux/rtl8723de #69: Build fails on new linux kernel 5.6.15 on arch linux

premraj59, 2020-05-31:
[image]

[make.log](https://github.com/smlinux/rtl8723de/files/4707655/make.log)

aideng1694, 2020-06-04:
Same issue on Fedora 32

Contents of `/var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/make.log`:
```
DKMS make.log for rtl8723de-5.1.1.8_21285.20171026_COEX20170111-1414 for kernel 5.6.15-300.fc32.x86_64 (x86_64)
Thu  4 Jun 14:11:28 BST 2020
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.6.15-300.fc32.x86_64/build M=/var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build  modules
make[1]: Entering directory '/usr/src/kernels/5.6.15-300.fc32.x86_64'
  CC [M]  /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/core/rtw_cmd.o
In file included from /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/include/drv_types.h:35,
                 from /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/core/rtw_cmd.c:22:
/var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/include/wifi.h:1006: warning: "IEEE80211_MAX_AMPDU_BUF" redefined
 1006 | #define IEEE80211_MAX_AMPDU_BUF 0x40
      | 
In file included from /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/include/osdep_service_linux.h:86,
                 from /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/include/osdep_service.h:42,
                 from /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/include/drv_types.h:32,
                 from /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/core/rtw_cmd.c:22:
./include/linux/ieee80211.h:1460: note: this is the location of the previous definition
 1460 | #define IEEE80211_MAX_AMPDU_BUF  0x100
      | 
  CC [M]  /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/core/rtw_security.o
In file included from /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/include/drv_types.h:35,
                 from /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/core/rtw_security.c:22:
/var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/include/wifi.h:1006: warning: "IEEE80211_MAX_AMPDU_BUF" redefined
 1006 | #define IEEE80211_MAX_AMPDU_BUF 0x40
      | 
In file included from /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/include/osdep_service_linux.h:86,
                 from /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/include/osdep_service.h:42,
                 from /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/include/drv_types.h:32,
                 from /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/core/rtw_security.c:22:
./include/linux/ieee80211.h:1460: note: this is the location of the previous definition
 1460 | #define IEEE80211_MAX_AMPDU_BUF  0x100
      | 
  CC [M]  /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/core/rtw_debug.o
In file included from /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/include/drv_types.h:35,
                 from /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/core/rtw_debug.c:22:
/var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/include/wifi.h:1006: warning: "IEEE80211_MAX_AMPDU_BUF" redefined
 1006 | #define IEEE80211_MAX_AMPDU_BUF 0x40
      | 
In file included from /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/include/osdep_service_linux.h:86,
                 from /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/include/osdep_service.h:42,
                 from /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/include/drv_types.h:32,
                 from /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/core/rtw_debug.c:22:
./include/linux/ieee80211.h:1460: note: this is the location of the previous definition
 1460 | #define IEEE80211_MAX_AMPDU_BUF  0x100
      | 
  CC [M]  /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/core/rtw_io.o
In file included from /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/include/drv_types.h:35,
                 from /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/core/rtw_io.c:52:
/var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/include/wifi.h:1006: warning: "IEEE80211_MAX_AMPDU_BUF" redefined
 1006 | #define IEEE80211_MAX_AMPDU_BUF 0x40
      | 
In file included from /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/include/osdep_service_linux.h:86,
                 from /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/include/osdep_service.h:42,
                 from /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/include/drv_types.h:32,
                 from /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/core/rtw_io.c:52:
./include/linux/ieee80211.h:1460: note: this is the location of the previous definition
 1460 | #define IEEE80211_MAX_AMPDU_BUF  0x100
      | 
  CC [M]  /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/core/rtw_ioctl_query.o
In file included from /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/include/drv_types.h:35,
                 from /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/core/rtw_ioctl_query.c:22:
[... cut ...]
   61 |             const struct proc_ops *,
      |             ^~~~~~~~~~~~~~~~~~~~~~~
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:268: /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/os_dep/linux/rtw_proc.o] Error 1
make[1]: *** [Makefile:1695: /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build] Error 2
make[1]: Leaving directory '/usr/src/kernels/5.6.15-300.fc32.x86_64'
make: *** [Makefile:1884: modules] Error 2
```

smlinux, 2020-06-06:
ed6ac722328ce24d6510f18daa0a7f7bc9d717d3

closed 2020-06-06 by smlinux