smlinux/rtl8723de #70: Bluetooth adapter RTL8723D is not getting detected in kernels after 4.19.6

ShubhamSharma64337, 2020-06-05:
Hi,
Whenever I install any Linux distro, if it has kernel version 4.19.0 or 4.19.6, Bluetooth works fine and I can see the detection of hci0 Bluetooth device in dmesg log and also the loading of firmware for my adapter Realtek rtl8723d succeeds. But, if the kernel version is 4.20 or 5.x or any newer one, Bluetooth adapter is not detected. I cannot see any hci0 device in the dmesg log or in rfkill. I have a dual boot setup and have Windows 10 installed alongside Linux. One more thing to notice is; suppose I am in the older kernel and Bluetooth is working fine, if I boot into the newer kernel once, and reboot into the older kernel, hci0 stops getting detected in the older kernels too! In order to start the detection of hci0 in older kernels again, I have to boot once into Windows and then boot into the older kernel, then hci0 starts getting detected again. It seems like the newer kernels are not only unable to detect Bluetooth, but also leave it in an unusable state for the older kernels, which strangely is solved only by Windows.

 

Here is the output of sudo dmesg | grep Blue on older kernels :

 

[ 12.264937] Bluetooth: Core ver 2.22

[ 12.264947] Bluetooth: HCI device and connection manager initialized

[ 12.264948] Bluetooth: HCI socket layer initialized

[ 12.264950] Bluetooth: L2CAP socket layer initialized

[ 12.264952] Bluetooth: SCO socket layer initialized

[ 12.402352] Bluetooth: hci0: RTL: rtl: examining hci_ver=08 hci_rev=000d lmp_ver=08 lmp_subver=8723

[ 12.403335] Bluetooth: hci0: RTL: rom_version status=0 version=2

[ 12.403336] Bluetooth: hci0: RTL: rtl: loading rtl_bt/rtl8723d_fw.bin

[ 12.440366] Bluetooth: hci0: RTL: rtl: loading rtl_bt/rtl8723d_config.bin

[ 18.871947] Bluetooth: BNEP (Ethernet Emulation) ver 1.3

[ 18.871947] Bluetooth: BNEP filters: protocol multicast

[ 18.871950] Bluetooth: BNEP socket layer initialized

[ 29.993005] Bluetooth: RFCOMM TTY layer initialized

[ 29.993010] Bluetooth: RFCOMM socket layer initialized

[ 29.993013] Bluetooth: RFCOMM ver 1.11

 

Here is the same command's output for newer kernels:

 

[ 95.362787] Bluetooth: Core ver 2.22

[ 95.362811] Bluetooth: HCI device and connection manager initialized

[ 95.362815] Bluetooth: HCI socket layer initialized

[ 95.362817] Bluetooth: L2CAP socket layer initialized

[ 95.362820] Bluetooth: SCO socket layer initialized

 

Is there any possible solution for this?

eduardoc25, 2020-06-06:
Hello
I use arch linux and since kernel 5.4.2 I have no problems with wi-fi or bluetooth ... using this driver.
Cheers
[image]

ShubhamSharma64337, 2020-06-07:
What should I do then?, wi-fi works fine even on newer kernels, the only problem is with Bluetooth.

eduardoc25, 2020-06-07:
First, make sure you don't have the @lwfinger module installed (the famous rtlwifi_new).
The module of that repository has this behavior, when you connect to a wi-fi network there is simply no way to make the bluetooth module work, going around and trying many things the one way I managed to get it going is not being connected to a network wi-fi, for some reason ... the controller seems to go crazy and disables the other ... just looking for a solution to this problem that I need to use wi-fi and bluetooth at the same time I found this repository ... and for me Surprise, not only can I use both at the same time, but the signal quality improves noticeably.

> [eduardo@RyzenR5 ~]$ sudo dmesg | grep Blue
[sudo] password for eduardo: 
[    4.050877] Bluetooth: Core ver 2.22
[    4.050900] Bluetooth: HCI device and connection manager initialized
[    4.050905] Bluetooth: HCI socket layer initialized
[    4.050908] Bluetooth: L2CAP socket layer initialized
[    4.050921] Bluetooth: SCO socket layer initialized
[    4.142452] Bluetooth: hci0: RTL: examining hci_ver=08 hci_rev=000d lmp_ver=08 lmp_subver=8723
[    4.145445] Bluetooth: hci0: RTL: rom_version status=0 version=2
[    4.145448] Bluetooth: hci0: RTL: loading rtl_bt/rtl8723d_fw.bin
[    4.146696] Bluetooth: hci0: RTL: loading rtl_bt/rtl8723d_config.bin
[    4.146884] Bluetooth: hci0: RTL: cfg_sz 10, total sz 33266
[    4.674489] Bluetooth: hci0: RTL: fw version 0x828a96f1
[    5.060669] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[    5.060685] Bluetooth: BNEP filters: protocol multicast
[    5.060689] Bluetooth: BNEP socket layer initialized
[   88.249049] Bluetooth: RFCOMM TTY layer initialized
[   88.249057] Bluetooth: RFCOMM socket layer initialized
[   88.249066] Bluetooth: RFCOMM ver 1.11

ShubhamSharma64337, 2020-06-08:
Can you please answer my few questions,
1. Do you have a dual boot system, if it is dual boot, what system is it?
2. Have you tried any other distros, if yes, did bluetooth work on them?

eduardoc25, 2020-06-08:
Do you have a dual boot system, if dual boot what system is it?
No, on the hp notebook that has the 8723de chip I only have Archlinux.
  
   Have you tried any other distribution? If yes, did Bluetooth work on them?
I have tested with opensuse tumbleweed on a sony notebook that I have (take out the hp and collect this one, for testing purposes, this one has dual boot in conjunction with W10), using the @ smlinux / rtl8723de module works for me correct way (both wi-fi and bluetooth at the same time) ...

eduardoc25, 2020-06-08:
These are the firmware that I see load when I raise the bluetooth and the wi-fi ... (I upload mine in case you don't have them)

[rtl8723de wifi and bluetooth.zip](https://github.com/smlinux/rtl8723de/files/4743602/rtl8723de.wifi.and.bluetooth.zip)

in both distros they are in the following directory:

- /lib/firmware/rtl_bt/rtl8723d_fw.bin
- /lib/firmware/rtl_bt/rtl8723d_config.bin
- /lib/firmware/rtlwifi/rtl8723defw.bin

Cheers

ShubhamSharma64337, 2020-06-08:
I also have all these files in their respective directories. I think the issue is just with the kernel of the distributions I have tried so far ( I have not used any of the distros which you have tried out ). I think I should just use either Debian or MX Linux which have older 4.19.x kernels. 

Thank you for the responses!