smlinux/rtl8723de #75: WIFI Error

hanu7674, 2020-07-16:
i have a hp laptop with Linux Os containing  rtl8723de wifi module.
On starting laptop when first time connecting to a network is fine but when connecting second time its not good ....its not connecting to that network but again restarting laptop its working help me in this issue

eduardoc25, 2020-07-16:
Hello, I think you should provide more information, or be more specific ... Details such as the distribution you use, the kernel version, what is the version of the module you installed, etc. will be welcome so they can provide help.
Regards

hanu7674, 2020-07-17:
OS distribution: Parrot os (Security)
Kernel version:5
debian
Module version as you posted.

imDaniX, 2020-07-21:
Same problem.
```
imdanix@GladOS:~$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 19.10
Release:	19.10
Codename:	eoan
imdanix@GladOS:~$ uname -a
Linux GladOS 5.3.0-62-generic #56-Ubuntu SMP Tue Jun 23 11:20:52 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux
```
Used method from [there](https://askubuntu.com/a/1235988).

imDaniX, 2020-07-22:
Just to clarify: WIFI networks are visible, but when you're trying to connect to the network, it connects, like, forever, and then fails. Also the Bluetooth connection works fine.

hanu7674, 2020-07-27:
> Just to clarify: WIFI networks are visible, but when you're trying to connect to the network, it connects, like, forever, and then fails. Also the Bluetooth connection works fine.

yes same as it is......

S4GAR, 2020-09-12:
Did you find the solution, Because I am also facing the same issue with my parrot 5.70-1 kernel update.

patchymoose, 2020-11-08:
Specifically, are you guys finding that this issue occurs after the laptop has been put to sleep? I find that the wifi works after a hard restart, but the issue occurs after it goes to sleep and then wakes back up.

I'm able to get the wifi to work by removing the module and inserting it again, but obviously that is a manual step every time you wake up the laptop.

But try: 

> sudo rmmod 8723de
> sudo modprobe rtl8723de

Seems to work for me, as simple/silly as that is...