smlinux/rtl8723de #80: No longer compatible with my kernel version

Wambug, 2020-09-18:
so yesterday after updating my system it suddenly told me that the rtl8723de has been uninstalled and if i would like to reinstall earlier versions i use this command -->dkms install rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414 so after trying reintsalling it gave me this error:
'make' all KVER=5.8.0-1parrot1-amd64................................(bad exit status: 2)
Error! Bad return status for module build on kernel: 5.8.0-1parrot1-amd64 (x86_64)
Consult /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/make.log for more information.

Fallast, 2020-09-19:
same issue here

Wambug, 2020-09-24:
> same issue here

i found a solution to this you have to remove the dkms module and rtl8723de module for the dkms i used----> sudo rm -r dkms command to do this you go to your lib folder open it on the terminal then used that command as for the rtl8723de i used ---->sudo rm -r rtl8723de
After removing everything I installed everything again and it worked

closed 2020-09-24 by Wambug

afranko, 2020-09-30:
I have exactly the same issue, but on Manjaro (5.7.15-1-MANJARO). Wambug's solution didn't work for me.

afranko, 2020-10-07:
> I have exactly the same issue, but on Manjaro (5.7.15-1-MANJARO). Wambug's solution didn't work for me.

In my case, the problem was the lack of compatibility between the kernel and GCC. The current kernel was compiled with GCC 10.0.1, but my GCC version was 10.0.2. So, I downgraded my GCC to 10.0.1 and now the driver works.