smlinux/rtl8723de #85: [ERROR] Unable to reconnect to any wifinetwork after disconnecting

anonymousguy198, 2020-10-27:
I'm unable to connect to any network after I disconnect. When I try to connect, It shows a loading icon and after that it fails. Only way of reconnecting is by rebooting. For connecting to a different network, I need to disable auto connect and reboot

OS = Ubuntu 18.04.5 LTS
kernel = 5.3.0-40-generic

Dmesg (after disconnecting)
`
[  508.175903] RTW: rtw_sta_mstatus_disc_rpt wlo1 - mac_id=0
[  508.175989] RTW: SetHwReg8723D(wlo1): [HW_VAR_MACID_WAKEUP] macid=0, org reg_0x4d4=0x00000000
[  508.175990] RTW: cfg80211_rtw_disconnect...call rtw_indicate_disconnect
[  508.175994] RTW: rtw_free_assoc_resources-wlo1 tgt_network MacAddress=1a:cf:c5:12:81:5dssid=Nokia 1
[  508.176001] RTW: ### Clean STA_(0) info ###
[  508.176004] RTW: free disconnecting network of scanned_queue
[  508.176010] RTW: rtw_cfg80211_unlink_bss(): cfg80211_unlink Nokia 1!!
[  508.176021] RTW: rtw_cfg80211_indicate_disconnect(wlo1)
[  508.176044] RTW: cfg80211_rtw_disconnect(wlo1) return 0
[  508.176050] RTW: rtw_ndev_notifier_call(wlo1) state:10
[  508.176052] RTW: rtw_reset_securitypriv(wlo1) - End to Disconnect
[  508.176285] RTW: netdev_close(wlo1) , bup=1
[  508.176287] RTW: (2)871x_drv - drv_close, bup=1, hw_init_completed=_TRUE
[  508.176290] RTW: -871x_drv - drv_close, bup=1
[  508.179823] RTW: rtw_ndev_notifier_call(wlo1) state:2
[  508.224566] RTW: cfg80211_rtw_get_txpower
[  508.307296] RTW: rtw_wx_get_rts, rts_thresh=2347
[  508.307299] RTW: rtw_wx_get_frag, frag_len=2346
[  512.659291] RTW: ==>rtw_ps_processor .fw_state(8)
[  512.659403] RTW: ==>ips_enter cnts:8
[  512.659404] RTW: nolinked power save enter
[  512.659406] RTW: ===> rtw_ips_pwr_down...................
[  512.659407] RTW: ====> rtw_ips_dev_unload...
[  512.661489] RTW: ====> rtl8723de_hal_deinit()
[  512.661493] RTW: =====>hal_poweroff_8723DE
[  512.674342] RTW: SetHwReg: bMacPwrCtrlOn=0
[  512.674345] RTW: rtl8723de_hal_deinit() <====
[  512.674348] RTW: <=== rtw_ips_pwr_down..................... in 12ms
[  580.036081] RTW: rtw_ndev_notifier_call(wlo1) state:14
[  580.036089] RTW: _netdev_open(wlo1) , bup=1
[  580.036099] RTW: -871x_drv - drv_open, bup=1
[  580.036207] RTW: cfg80211_rtw_set_power_mgmt(wlo1) enabled:1, timeout:-1
[  580.036243] RTW: rtw_ndev_notifier_call(wlo1) state:1
[  580.068728] RTW: cfg80211_rtw_get_txpower
[  580.068754] RTW: cfg80211_rtw_get_txpower
[  580.068883] RTW: cfg80211_rtw_get_txpower
[  580.068898] RTW: cfg80211_rtw_get_txpower
[  580.069186] RTW: cfg80211_rtw_get_txpower
[  580.069219] RTW: rtw_ndev_notifier_call(wlo1) state:4
[  580.096387] RTW: cfg80211_rtw_del_key(wlo1) key_index=0
[  580.096409] RTW: cfg80211_rtw_del_key(wlo1) key_index=1
[  580.096445] RTW: cfg80211_rtw_del_key(wlo1) key_index=2
[  580.096468] RTW: cfg80211_rtw_del_key(wlo1) key_index=3
[  580.096479] RTW: cfg80211_rtw_del_key(wlo1) key_index=4
[  580.096490] RTW: cfg80211_rtw_del_key(wlo1) key_index=5
[  580.096499] RTW: cfg80211_rtw_flush_pmksa(wlo1)
[  580.123868] RTW: rtw_ndev_notifier_call(wlo1) state:4
[  580.125004] RTW: cfg80211_rtw_scan(wlo1)
[  580.125008] RTW: _rtw_pwr_wakeup call ips_leave....
[  580.125009] RTW: ==>ips_leave cnts:8
[  580.125010] RTW: ===>  rtw_ips_pwr_up..............
[  580.125014] RTW: ===> ips_netdrv_open.........
[  580.125015] RTW: =======>rtl8723de_hal_init()
[  580.125093] RTW:  power-on :REG_CR 0x100=0xea.
[  580.125099] RTW: =======>InitMAC_8723D()
[  580.129666] RTW: SetHwReg: bMacPwrCtrlOn=1
[  580.129860] RTW: InitMAC_8723D() <====
[  580.129869] RTW: rtl8723d_FirmwareDownload(wlo1) tmp_ps=0
[  580.129870] RTW: rtl8723d_FirmwareDownload fw: FW_NIC, size: 26498
[  580.129872] RTW: rtl8723d_FirmwareDownload: fw_ver=1d fw_subver=0002 sig=0x23d1, Month=04, Date=14, Hour=11, Minute=13
[  580.129872] RTW: rtl8723d_FirmwareDownload(): Shift for fw header!
[  580.129875] RTW: rtl8723d_FirmwareDownload by IO write!
[  580.132405] RTW: polling_fwdl_chksum: Checksum report OK! (1, 0ms), REG_MCUFWDL:0x00060705
[... 502 lines cut ...]
[  608.757917] RTW:  <=== rtl8723d_FirmwareDownload()
[  608.757917] RTW: FwLoad SUCCESSFULLY!!!
[  608.757918] RTW: 8723D MAC Config Start!
[  608.757941] RTW: 8723D MAC Config Finished!
[  608.757941] RTW: BB Config Start!
[  608.758633] RTW: BB Config Finished!
[  608.758634] RTW: RF Config started!
[  608.817267] RTW: RF Config Finished!
[  608.818104] RTW: ===>phy_SpurCalibration_8723D: Channel = 1
[  608.818115] RTW: ===>phy_SwChnl8723D: Channel = 1
[  608.818195] RTW: InitializeAdapter8723DE(): Turn on  ----------
[  609.095036] RTW: rtw_hal_set_macaddr_port wlo1- hw port(0) mac_addr =fc:01:7c:9a:3c:0b
[  609.095051] RTW: rtw_hal_get_macaddr_port wlo1- hw port(0) mac_addr =fc:01:7c:9a:3c:0b
[  609.095052] RTW: wlo1- hw port(0) mac_addr =fc:01:7c:9a:3c:0b
[  609.095473] RTW: ===>phy_SpurCalibration_8723D: Channel = 3
[  609.095484] RTW: ===>phy_SwChnl8723D: Channel = 3
[  609.095557] RTW: rtw_hal_set_macaddr_port wlo1- hw port(0) mac_addr =fc:01:7c:9a:3c:0b
[  609.095564] RTW: #### hw_var_set_opmode() -4657 hw_port(0) mode = 2 ####
[  609.095566] RTW: +pci_intf_start
[  609.095567] RTW: -pci_intf_start
[  609.095568] RTW: <===  rtw_ips_pwr_up.............. in 364ms
[  609.095569] RTW: nolinked power save leave
[  609.095572] RTW: ==> ips_leave.....LED(0x08828282)...
[  609.095859] RTW: cfg80211_rtw_disconnect(wlo1) return 0
`

efeyaryukey75, 2020-12-25:
Same problem on my Debian based Pardus Linux. When i lost my connection i have to reboot my laptop to reconnect. 

OS: Pardus 19.4
Kernel Version: 5.9 bpo

vi88i, 2021-02-07:
Anyone found a quick fix for this issue?