smlinux/rtl8723de #87: Bug on compilation with Linux 5.10

FlowArg, 2020-12-31:
Hello,
I have a problem when I run `sudo dkms install --no-depmod -m rtl8723de-5.1.1.8_21285.20171026_COEX20170111 -v 1414 -k 5.10.2-2-MANJARO` with the last version of your module.
Here is my logs : https://code.alwyn974.re/qeqosexafe.log

And I read this reddit : https://www.reddit.com/r/archlinux/comments/knuj47/potential_dkms_compile_errors_with_kernel_510/

Is there possible to solve this problem ?

Thanks so much for this module very helpful everyday.

My specs :
**Linux Distribution:** Manjaro (latest version)
**Kernel:** Linux kernel 5.10.2
**DE:** KDE Plasma 5.20.4

Vadim-Burns, 2021-02-06:
Hello, i have solved this problem by switching from current branch to 5.10-up. Have u tried this?

FlowArg, 2021-02-06:
Hello, there is no "5.10-up" branch, you mean the "5.0-up" branch?

kelppo, 2021-02-09:
You can just use the upstream driver instead, it's named rtw88_8723de, if it's not available in your distro kernel ask the maintainers to enable it.
https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/?id=f5df1a8b4376551f7fb2416135c58896b70a1467

FlowArg, 2021-02-09:
I haven't had any problems since I switched to Arch Linux. The driver is now in the kernel as you said. Thank you!

closed 2021-02-09 by FlowArg